indilib/indi-3rdparty #2: gphoto shutter speed mismatch

herrhaus, 2019-09-18:
Reproduce:
1) Connect DSLR in "M" mode
2) set "force bulb" to OFF
3) In CCD TAB add a 1/1000s exposure to the queue
4) activate the queue
5) Picture taken is a 1/500s

Expected behaviour would be a 1/1000s picture taken.

Logfile: 

[2019-09-17T13:11:53.645 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 5D Mark II : "[DEBUG] Finding optimal exposure setting for 0.001001 seconds in shutterspeed (count=37)... "
[2019-09-17T13:11:53.645 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 5D Mark II : "[DEBUG] Closest match: 0.002 seconds Index: 28 "
[2019-09-17T13:11:53.645 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 5D Mark II : "[DEBUG] Setting radio/menu widget shutterspeed: 28 (1/500) "
[2019-09-17T13:11:53.672 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 5D Mark II : "[DEBUG] Setting new configuration OK. "
[2019-09-17T13:11:53.672 CEST DEBG ][           org.kde.kstars.indi] - Canon DSLR EOS 5D Mark II : "[DEBUG] Using predefined exposure time: 0.002 seconds "

Ekos considers 0.001001 closer to 0.002 than 0.001 thus selects 1/500s.

Similar behaviour for other very short shutter speeds (Ekos chooses 1/1000 when 1/8000 was selected).

Kubuntu 19.04, KStars Build: 2019-09-18T13:06:05Z. Camera used: Canon 5D Mk2. Complete logfile see: https://indilib.org/forum/ccds-dslrs/5622-eos-1-4000-shuuter-speed.html?start=12#43479

tstibor, 2019-09-18:
See PR https://github.com/indilib/indi-3rdparty/pull/4

tstibor, 2019-09-18:
Probably if (std::fabs(delta) < 1e-6) is too strict and 1e-4 is better suited, one need to test that.

closed 2019-09-18 by knro

knro, 2019-09-18:
it's 1e-5 now, please reopen if it is still problematic.

herrhaus, 2019-09-18:
Works, thanks a lot!

commit 500e3e57 (2019-09-18): Make delta check tighter, 1e-5 seems to correctly match all up to 1/8000.

Fixes #2