indilib/indi-3rdparty #7: SONY A55 not working with SONY DLSR driver.

philpugh51, 2019-09-21:
This issue being reported in here at request of Jasem..

I have a SONY A55 which is not working with the SONY DLSR driver.  I have attached two verbose logs of the event but the nub of the issue seems to be that the driver does not detect that the camera is in BULB mode - get a 'cannot find exposure widget' error.

FWIW This is also reporduceableon my NEX-6 and NEX-7 bodies - but I understand these have a different firmware structure to the Alpha models.  Getting either to work would be great.

[log_18-54-36.txt](https://github.com/indilib/indi-3rdparty/files/3638604/log_18-54-36.txt)
[log_11-16-45.txt](https://github.com/indilib/indi-3rdparty/files/3638605/log_11-16-45.txt)

knro, 2019-09-22:
Not sure if I asked you this before, but can you capture with gphoto2?

`gphoto2 --debug --debug-logfile=gphoto.txt --capture-image-and-download`

karlrees, 2019-11-05:
Phil- I have the A55 too.  I thought I had researched it pretty thoroughly and determined that it wasn't possible to support it.  For instance, it's not listed here: http://www.gphoto.org/proj/libgphoto2/support.php.  How are you trying to connect it?  I'd be happy to find out I was wrong, but I think it's just too old of a camera.

philpugh51, 2019-11-05:
**karlrees** 

I have come to the same conclusion and have given up the idea of using any of my three Sony cameras (A55, NEX-6 and NEX-7) for astrophotography !

closed 2020-04-29 by knro