indilib/indi-3rdparty #21: indi-toupbase source build make error

Fire-Power, 2019-11-09:
When building the indi-toupbase 3rd party driver I get an error during the make.
I have built several other drivers in the same environment without issue today.
Raspbian Buster on pi 3B+

`pi@pi4:~/codedev/indi-3rdparty/build/indi-toupbase $ make
[ 12%] Building CXX object CMakeFiles/indi_altair_ccd.dir/indi_toupbase.cpp.o
/home/pi/codedev/indi-3rdparty/indi-toupbase/indi_toupbase.cpp: In member function ‘void ToupBase::pushCallback(const void*, const AltaircamFrameInfoV2*, int)’:
/home/pi/codedev/indi-3rdparty/indi-toupbase/indi_toupbase.cpp:2108:22: error: ‘unique_lock’ is not a member of ‘std’
                 std::unique_lock<std::mutex> guard(ccdBufferLock); `

Could you give me any pointers as to what might be wrong please.

knro, 2019-11-10:
make sure you installed latest libindi first.

Fire-Power, 2019-11-10:
Hi 
I have cloned and built a complete new virgin Indi  installation from here a few days ago.
I have successfully built EQMOD, and GPSD drivers and also a Motor Hat Focuser from another github site, all from this install without any issues.  So unless I have broken something without realising it, I believe it is fully up to date.

Fire-Power, 2019-11-11:
I found out where I went wrong.
I build libindi with the command       git clone --depth 1
It obviously needs a full install for this driver to build.
Thank you for the help.
And thank you for such a great project.

closed 2019-11-11 by Fire-Power