indilib/indi-3rdparty #24: ldconfig returns "is truncated" for latest sbig libs

rkaczorek, 2019-11-14:
ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so.4 is truncated
ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so is truncated
ldconfig: file /usr/lib/arm-linux-gnueabihf/libsbig.so.4.9.9 is truncated

openastroproject, 2019-11-14:
I have also seen this issue on 64-bit Intel Mint 19

knro, 2019-11-14:
So while the version is 4.9.9, sbig internal name is still libsbigudrv.so.1

So I used patchelf (latest version) to change the soname to libsbig.so.4. This was done on 18.04 and it appears to be working on Ubuntu. SBIG can be *extremely* slow in responding to changes so I went ahead with the patchelf. But it would great if they can release a fixed version.

Another issue I found is that their armhf version was compiled under buster and requires glibc 2.28 which is not available on most LTS distros now. I again asked them to update it and awaiting their response. Any suggestions welcome.

elbarbudo, 2020-04-01:
I have also seen this issue on raspberry PI 3 Raspbian Buster

ddanno, 2020-04-10:
I too encountered this issue last night on Debian Buster.  This is related to https://sourceware.org/bugzilla/show_bug.cgi?id=23964 and https://sourceware.org/bugzilla/show_bug.cgi?id=25087.  The libsbig version in the latest INDI had its soname set to libsbig.so.4 using patchelf, as Jasem mentioned in a prior post.  After a bit of digging, I located this thread: https://forum.diffractionlimited.com/threads/linux-drivers-for-arm-v8-64bit.6290/
that contained a link to a 64-bit Debian package:
https://forum.diffractionlimited.com/attachments/sbigudrv_4-99_amd64-zip.6432/

Downloads of this package are restricted to Diffraction Limited forum members;  don't ask why.  I created an account and downloaded and installed the package:

# dpkg -l libsbigudrv1
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libsbigudrv1   4.99-1       amd64        SBIG Universal Driver Linux SDK
# dpkg -L libsbigudrv1
/.
/lib
/lib/firmware
/lib/firmware/sbigfcam.hex
/lib/firmware/sbiglcam.hex
/lib/firmware/sbigpcam.hex
/lib/firmware/sbigucam.hex
/lib/firmware/stfga.bin
/lib/udev
/lib/udev/rules.d
/lib/udev/rules.d/51-sbig-debian.rules
/usr
/usr/include
/usr/include/sbigudrv.h
/usr/lib
/usr/lib/libsbigudrv.so.1
/usr/share
/usr/share/doc
/usr/share/doc/libsbigudrv1
/usr/share/doc/libsbigudrv1/changelog.Debian.gz
/usr/share/doc/libsbigudrv1/copyright
/usr/lib/libsbigudrv.so

Crucially, ldconfig is happy with this /usr/lib/libsbigudrv.so;  it does not consider it truncated:
# ls -l /usr/lib/libsbigudrv*
lrwxrwxrwx 1 root root     16 Oct 28 07:02 /usr/lib/libsbigudrv.so -> libsbigudrv.so.1
-rw-r--r-- 1 root root 280624 Oct 28 07:02 /usr/lib/libsbigudrv.so.1
# ldconfig -l /usr/lib/libsbigudrv.so.1
#

The last step was to recompile indi-sbig (out of the indi-3rdparty git repo) linking against libsbigudrv.  This required editing indi-3rdparty/cmake_modules/FindSBIG.cmake, replacing "find_library(SBIG_LIBRARIES NAMES sbig" with "find_library(SBIG_LIBRARIES NAMES sbigudrv".

I still do not know where SBIG/Diffraction Limited's official repository resides.  ftp://sbig.com/pub/ looked promising and contains some useful information, but not the Linux driver I was looking for.  If anyone could point me in the right direction, I'd appreciate it.

knro, 2021-09-28:
Closing as this must be fixed by upstream SBIG.

closed 2021-09-28 by knro