indilib/indi-3rdparty #39: Support for SNAP port on EQMOD Mounts

setnes, 2019-12-22:
This has been previously discussed here.  -->  https://github.com/indilib/indi/issues/617

I would love to be able to use the SNAP port on my EQ6-R Pro via EQMOD.  Is there a way to do this yet... even via a simple script?

geehalel, 2020-03-29:
Hi,
I had already implemented that feature in my own fork. I just remake a fork and put it again in a snapport branch. As you notice, I was using a small script to monitor the opening of a file by the gphoto driver.  I wonder if, snooping a ccd device exposure property, the eqmod driver may toggle a snapport when the exposure property turns idle (the ccd is in exposure). Or maybe there could be other usages...

setnes, 2020-03-30:
I built the code in your snapport branch, but I don't see any new INDI parameters for the snap ports.  I'm not sure what I should expect to see though.  I have an EQ6-R Pro.

setnes, 2020-03-30:
I put a "return true;" inside the HasSnapPort1 function and was able to make it work.  I plugged a camera (Fujifilm X-T20) into the snap port and was able to open and close the shutter using the INDI parameter.  So... the next thing you'll want is a mount code.  

`22:43:20.100: [DEBUG] InquireBoardVersion(): MountCode = 35, MCVersion = 30723, setting minperiods Axis1=6 Axis2=6`

Should we assume that the 35 printed in the log should be a 0x23 in the code?  Let me try that.

geehalel, 2020-03-30:
Thank you for testing, and solving the issue with EQ6-R (I did'nt know the mount code and did not want to look at the firmware). I added the mount in my repo so it should be ok now. I've just tested
with the simulator giving a 0x23 mount code. Maybe could you test again before I make a PR.
Thanks again.

setnes, 2020-03-30:
There might be two mount codes for EQ6-R Pro.  There are different firmware downloads for mounts with USB and mounts without.  I have a newer one that has USB.  I know someone with the older version of the mount, but we have stay-at-home orders right now for COVID-19.  I'm not sure if I'd be able to step him through finding the mount code remotely.

I will test your new version, but cannot do it until after work.  I also did not try your script (from your older branch) to make this available to gphoto.  Instead I tested by using indi_setprop to open and close the shutter from a simple bash script.

geehalel, 2020-03-30:
Ok, you're right, there is also a firmware V2.11 for AZEQ6 and EQ6-R (not pro?) mounts on the
skywatcher download page. Let's wait and know more about that later.
I also commented about the polar scope led issue and wonder if you get a notification from github as the issue is now closed.

setnes, 2020-03-30:
I cannot build your changed branch.  This looks like it is related to "Use updated API call" changes @knro was making.

`/home/astroberry/Projects/indi-3rdparty/indi-eqmod/eqmodbase.cpp: In member function ‘virtual bool EQMod::Handshake()’:
/home/astroberry/Projects/indi-3rdparty/indi-eqmod/eqmodbase.cpp:718:13: error: ‘tty_set_generic_udp_format’ was not declared in this scope
             tty_set_generic_udp_format(1);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-eqmod/eqmodbase.cpp:718:13: note: suggested alternative: ‘tty_set_gemini_udp_format’
             tty_set_generic_udp_format(1);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~
             tty_set_gemini_udp_format
`

geehalel, 2020-03-31:
Ah yes, I rebase my fork to stay in sync with the upstream indi-3rdparty repo, but I also build the 
indi-core from github so that works for my setup. We may wait astroberry to upgrade.

setnes, 2020-03-31:
If I rebuilt indi-core, would I run into issues with the other 3rd party drivers I am using as well... not just indi-eqmod?  

I *might* be able to try rebuilding core tonight.  It would be nice to get this tested before the next Astroberry release so I can eventually get back to a vanilla install.  I think I'd start with a full backup of this SD card before I try.  I'm worried I will create a mess.  :)

geehalel, 2020-03-31:
Concerning other third party drivers, I don't know without looking at every commits, but it's almost 
certain that you would have to rebuild them against the new indi-core.
I believe the simplest way would be to just 'revert' by hand the API call change: replace the 
`tty_set_generic_udp_format(1); ` with `tty_set_skywatcher_udp_format(1);` in `eqmodbase.cpp`
as it was before. That's the only change in the commit Jasem made on sunday.

setnes, 2020-04-01:
I tested your snapport branch.  I only made the one tweak to the udp_format line mentioned above (not related to this change).  My EQ6-R Pro mount was identified as one that supported snap port 1.  I was able to open and close the shutter on my camera using the INDI control panel via Cartes du Ciel.  I also tested opening and closing the shutter with indi_setprop at the command line.  Everything still works as expected.  👍 

This change is very helpful for people like me with cameras that do not support tethered shooting over USB.  I can write simple scripts to control the shutter on my camera using the INDI properties. 

I looked at your eqsnap1.py script from your old branch.  I don't think this would be helpful for me, but might be helpful for others.  I need to control the shutter on my camera without extracting images from the camera.   My camera shutter cannot be fired, in any way, when a USB connection exists.  I know some cameras allow for this, and your script would allow the GPhoto CCD driver to control the shutter via the snap port while still downloading photos over USB.  That would be handy for someone with such a camera.  I'm unable to test that though.  :)

Thanks again for your work on this.  It's very awesome.

geehalel, 2020-04-01:
Ok, many thanks to take the time to perform these tests. I will make the PR now.
Concerning your camera setup, I have not thought to your situation. I think I will leave the driver as is
and won't overload it with snooping stuff to control the snap ports. Only very few users still use
a 350D with INDI.

rkaczorek, 2020-04-18:
> Ah yes, I rebase my fork to stay in sync with the upstream indi-3rdparty repo, but I also build the
> indi-core from github so that works for my setup. We may wait astroberry to upgrade.

Just to let you know - astroberry has been upgraded to the latest INDI core libs

setnes, 2020-04-20:
Thanks @rkaczorek!

closed 2020-04-22 by knro