indilib/indi-3rdparty #44: indi-qhy compilation error

JBielanski, 2020-01-04:
**Describe the bug**
indi_qhy_ccd didn't compile, I received compilation error.

**To Reproduce**
1. Install indi-1.8.3
2. Build and install libqhy from indi-3rdparty-1.8.3
3. Try build indi-qhy
3a. mkdir BUILD && cd BUILD in  indi-3rdparty-1.8.3 directory
3b. cmake -DCMAKE_INSTALL_PREFIX=/usr ../indi-qhy/
3c. make

**Expected behavior**
A compilation bug appear in indi-3rdparty-1.8.3 version, with all previous versions is no problem.
Compilation process stopped with error:
error: 'SetQHYCCDBufferNumber' was not declared in this scope
             SetQHYCCDBufferNumber(USBBufferN[0].value);

**Desktop (please complete the following information):**
 - OS: Linux Enceladus 5.4.7-gentoo #1 SMP Wed Jan 1 21:32:43 CET 2020 x86_64 Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz GenuineIntel GNU/Linux
 - GCC: gcc version 8.3.0 (Gentoo 8.3.0-r1 p1.1)
 - CMake: cmake version 3.14.6
 - Version INDI: 1.8.3

**Log Files**
-------------- CMAKE LOG:
cmake -DCMAKE_INSTALL_PREFIX=/usr ../indi-qhy/
-- The CXX compiler identification is GNU 8.3.0
-- The C compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found CFITSIO 3.41: /usr/lib64/libcfitsio.so
-- Found INDI: /usr/lib64/libindidriver.so;/usr/lib64/libindiAlignmentDriver.so (found version "1.8.3") 
-- Found QHY: /usr/local/lib64/libqhyccd.so
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.11") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.3") 
-- Found USB1: /usr/lib64/libusb-1.0.so (found version "1.0.21") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jbielanski/Dokumenty/PROGRAMY/INDI-Astronomical-Suite/INDI/indi-3rdparty-1.8.3/BUILD
-------------- CMAKE LOG:

-------------- COMPILATION LOG:
make
Scanning dependencies of target qhy_ccd_test
[ 16%] Building CXX object CMakeFiles/qhy_ccd_test.dir/qhy_ccd_test.cpp.o
[ 33%] Linking CXX executable qhy_ccd_test
[ 33%] Built target qhy_ccd_test
Scanning dependencies of target indi_qhy_ccd
[ 50%] Building CXX object CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o
/home/jbielanski/Dokumenty/PROGRAMY/INDI-Astronomical-Suite/INDI/indi-3rdparty-1.8.3/indi-qhy/qhy_ccd.cpp: In member function 'virtual bool QHYCCD::ISNewNumber(const char*, const char*, double*, char**, int)':
/home/jbielanski/Dokumenty/PROGRAMY/INDI-Astronomical-Suite/INDI/indi-3rdparty-1.8.3/indi-qhy/qhy_ccd.cpp:1617:13: error: 'SetQHYCCDBufferNumber' was not declared in this scope
             SetQHYCCDBufferNumber(USBBufferN[0].value);
             ^~~~~~~~~~~~~~~~~~~~~
/home/jbielanski/Dokumenty/PROGRAMY/INDI-Astronomical-Suite/INDI/indi-3rdparty-1.8.3/indi-qhy/qhy_ccd.cpp:1617:13: note: suggested alternative: 'SetQHYCCDBurstIDLE'
             SetQHYCCDBufferNumber(USBBufferN[0].value);
             ^~~~~~~~~~~~~~~~~~~~~
             SetQHYCCDBurstIDLE
make[2]: *** [CMakeFiles/indi_qhy_ccd.dir/build.make:63: CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/indi_qhy_ccd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

-------------- COMPILATION LOG:

knro, 2020-01-05:
Please make sure to install libqhy to the system first before compiling indi-qhy

JBielanski, 2020-01-05:
I recompiled **libqhy** from indi-3rdparty-1.8.3, now it is looking that after installation:
> lrwxrwxrwx 1 root root      14 01-05 13:46 /usr/lib64/libqhyccd.so -> libqhyccd.so.6
> lrwxrwxrwx 1 root root      18 01-05 13:46 /usr/lib64/libqhyccd.so.6 -> libqhyccd.so.6.0.5
> -rw-r--r-- 1 root root 8709840 01-05 13:46 /usr/lib64/libqhyccd.so.6.0.5
CMake detect correct library:
> -- Found QHY: /usr/lib64/libqhyccd.so
But still I received the same error:  
> error: ‘SetQHYCCDBufferNumber’ was not declared in this scope
>             SetQHYCCDBufferNumber(USBBufferN[0].value);

JBielanski, 2020-01-05:
Problem solved! I found two headers files in system and one of them is wrong (probably not deleted in previous INDI update).

closed 2020-01-05 by JBielanski