indilib/indi-3rdparty #46: cmake fails if no FFMPEG library is installed

sterne-jaeger, 2020-01-08:
make fails for indi-3rdparty if no FFMPEG libraries are installed. Here is the matching output of cmake:

```
CMake Error at cmake_modules/FindFFmpeg.cmake:23 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/x86_64-linux-gnu/libavdevice/version.h
Call Stack (most recent call first):
  cmake_modules/FindFFmpeg.cmake:100 (_FFMPEG_PACKAGE_check_version)
  CMakeLists.txt:107 (find_package)

CMake Error at cmake_modules/FindFFmpeg.cmake:23 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/x86_64-linux-gnu/libavformat/version.h
Call Stack (most recent call first):
  cmake_modules/FindFFmpeg.cmake:117 (_FFMPEG_PACKAGE_check_version)
  CMakeLists.txt:107 (find_package)

CMake Error at cmake_modules/FindFFmpeg.cmake:23 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/x86_64-linux-gnu/libavutil/version.h
Call Stack (most recent call first):
  cmake_modules/FindFFmpeg.cmake:134 (_FFMPEG_PACKAGE_check_version)
  CMakeLists.txt:107 (find_package)

CMake Error at cmake_modules/FindFFmpeg.cmake:23 (file):
  file failed to open for reading (No such file or directory):

    /usr/include/x86_64-linux-gnu/libswscale/version.h
Call Stack (most recent call first):
  cmake_modules/FindFFmpeg.cmake:151 (_FFMPEG_PACKAGE_check_version)
  CMakeLists.txt:107 (find_package)

-- Found FFMPEG: /usr/lib/x86_64-linux-gnu/libavcodec.so;/usr/lib/x86_64-linux-gnu/libavdevice.so;/usr/lib/x86_64-linux-gnu/libavformat.so;/usr/lib/x86_64-linux-gnu/libavutil.so;/usr/lib/x86_64-linux-gnu/libswscale.so, /usr/include/x86_64-linux-gnu
-- Since FFMPEG was found, INDI Webcam Driver can be built
```
And later on
```
-- Configuring incomplete, errors occurred!
```

As soon as I install ```libavcodec-dev``` and ```libavdevice-dev```, the build succeeds.

knro, 2020-01-09:
@rlancaste Can you please check what needs to be done for INDI Webcam driver?

knro, 2020-03-15:
@rlancaste IIRC you solved this issue, correct?

closed 2020-04-29 by knro