indilib/indi-3rdparty #57: indi_atik_ccd refuses connecting on ubuntu 19.xx

spektros, 2020-02-07:
My Atik 460 does not connect to indiserver on Unbuntu 19.xx, but on Raspi Buster and on Artemis,Win7
Lodestar and QHY5 are working erverywhere.

:~$ indiserver indi_atik_ccd
2020-02-07T08:45:56: startup: indiserver indi_atik_ccd 
Child process 17635 died
2020-02-07T08:45:57: Driver indi_atik_ccd: stderr EOF
<delProperty device="Atik 460ex"/>
2020-02-07T08:45:57: Driver indi_atik_ccd: restart #1
Child process 17651 died
...
020-02-07T08:46:07: Driver indi_atik_ccd: restart #10
Child process 17797 died
<delProperty device="Atik 460ex"/>
2020-02-07T08:46:08: Driver indi_atik_ccd: Terminated after #10 restarts.
2020-02-07T08:46:08: good bye

OS:
Ubuntu 19.10
fresh install off indi-full from ppa 2020-02-07:12:04:73

openastroproject, 2020-02-07:
I may have the same problem with an Atik 314L+ on Mint19.  Certainly indi_atik_ccd seems to fail, and rolling back to the previous version I had installed works.  I'll try to provide more information about this over the weekend.

knro, 2020-02-07:
Did you reinstall libatik again?

spektros, 2020-02-07:
apt search libatik:
libatik/eon,now 2.2.6~202001312249~ubuntu19.10.1

knro, 2020-03-15:
Is this still an issue? Please report back.

spektros, 2020-03-15:
This problem is no longer existent.
See
https://indilib.org/forum/ccds-dslrs/6412-atik-460-not-working.html#49398

closed 2020-03-15 by spektros