indilib/indi-3rdparty #60: Cooler on/off control for Moravian Instruments cameras

sergio-dr, 2020-02-26:
@sWski In Ekos/Capture module, the cooler On/Off buttons are greyed out when using my MI G2-8300FW (anyway, I can set the desired temperature in the capture module or indi panel, and by doing this the cooler actually does its job). 

But the on/off capability is interesting when using Ekos scheduler, so when jobs end, it can automatically warm up the ccd (now I have to manually set the ambient temperature value to force this when I wake up). 

I tried to track down this and it seems that the libgxccd library doesn't implement this on/off control (I may be wrong, though). Could this be done? If this is the case, then it seems that indi-mi should implement the property CCD_COOLER for the Ekos On/Off buttons to enable. 

Thanks in advance. I understand this is secondary, just wanted to note that it would improve the experience using MI cameras with the Ekos scheduler.

sWski, 2020-02-26:
I've discussed this with @knro via email. Gx cameras with cooler doesn't have any way to _turn off_ the cooler. The only way to stop cooling the chip is to set the required CCD temperature at or above ambient temperature and the ON/OFF control should exactly do that. That's why there isn't any such function in the header file.

I can fix this in upcoming release, which will add support for new CMOS cameras.

sergio-dr, 2020-02-26:
Thanks Jakub. No hurry, if it can make it for an upcoming release at all, it'll be great. 
I've just discovered (facepalm) I can workaround this by invoking e.g.:
`/usr/bin/indi_setprop "MI CCD G2-8300.CCD_TEMPERATURE.CCD_TEMPERATURE_VALUE=25.0"`
inside a (scheduler) shutdown script.

knro, 2020-03-15:
@sWski That would be great, let me know when you're ready to merge this in!

knro, 2020-04-22:
@sWski was just about this today... is there an ETA when this would be available?

sWski, 2020-04-30:
@sergio-dr I've added the cooler ON/OFF switch in #113, please let me know if everything is ok. Note that the fan will be always on, only the cooling will be turned off and cooling power will be 0%.

PR #113 (2020-04-30): Add new CMOS cameras
- adds support for all new CMOS cameras from C1, C2 and C1+ series
- closes #60
closed 2020-04-30 by knro