indilib/indi-3rdparty #70: Support for polar scope LED brightness on EQMod Mounts.

setnes, 2020-03-14:
**Is your feature request related to a problem? Please describe.**
I have an EQ6-R Pro.  I have not found a way to set the polar scope LED brightness without using the hand controller.  If I set the brightness with the hand controller, and then use EQMod instead of the hand controller, the brightness reverts back to 100%.  It is not saved.  This makes it very hard to see Polaris.

**Describe the solution you'd like**
I would like an INDI parameter to set the brightness level of the polar scope LED.  This would eliminate the need to have the hand controller connected when using the polar scope.

knro, 2020-03-15:
Can you investigate if there is a command set for this?

@geehalel Any ideas about this?

setnes, 2020-03-15:
I looked in the [Motor Controller Command Set](https://skywatcher.com/download/manual/application-development/) document for the first time just now.  It is listed as "Set Polar Scope LED brightness."  I haven't looked at this long enough to know what any of the stuff on the command set table means though.  :)

There is also an [EQDimmer](https://sourceforge.net/projects/eq-mod/files/EQDIMMER/) utility for the ASCOM world.

closed 2020-03-16 by knro

knro, 2020-03-16:
Please test and report back.

setnes, 2020-03-17:
Awesome!  I will need to figure out how to build and test this.  Be patient.  :)

setnes, 2020-03-17:
It works!  Thanks.  :)

I decided to test on a Gentoo server instead of trying to build this on a Raspberry Pi (Astroberry)... it probably would have been easier to build it on the Raspberry Pi.  I ended up needing to recompile my kernel to support the USB to Serial device in the mount before I could see it work. 

Anyway, this is perfect.  I didn't do any regression tests though.  I only made sure the LED changed when I changed the INDI property.

geehalel, 2020-03-29:
Hi,
Sorry but did not have time to look at this before. Thanks anyway for implementing it.
There is a bit in the features which indicates support for polar scope led brightness adjustment.
It is already in the code, so I wonder if we should enable this feature only for those mounts.
Actually we should, the driver throws an exception as it is unsupported by the firmware on my AZEQ5.

geehalel, 2020-03-29:
I added the code to check if mount supports led brightness adjustment in my ledcheck branch.
It supposes the mount has the GetFeatureCmd in the firmware. But I don't know if every mounts
supporting led adjustment have also this GetFeatureCmd (EQ6 ?). So should I do a PR ?

setnes, 2020-03-30:
I'll be happy to verify this still works on my mount after your change, but right now it doesn't build for the same reasons the snapport branch doesn't build.

`eqmodbase.cpp:704:13: error: ‘tty_set_generic_udp_format’ was not declared in this scope`

setnes, 2020-04-01:
@geehalel, I tested your ledcheck branch with my EQ6-R Pro.  The LED feature is still working with your change that checks for the LED feature.

I did tweak unrelated code to avoid rebuilding indi-core as discussed [here](https://github.com/indilib/indi-3rdparty/issues/39#issuecomment-606669645).

geehalel, 2020-04-01:
Ok, thanks for testing. So I will keep the version which performs the test using the getFeatureCmd.
We may add later other mounts if they have polar scope led control.

commit 9379f0b7 (2020-03-16): Add LED brightness control

Fixes #70