indilib/indi-3rdparty #78: Unsupported architecture: aarch64

gnthibault, 2020-03-28:
I was not able to build indi-3rdparty on commit: 

> c18244dec950bde3aa2660194a9a373a1368c364

Build error:
```
CMake Error at libricohcamerasdk/CMakeLists.txt:26 (message):
  Unsupported architecture: aarch64
```

Thank you in advance for your help

knro, 2020-03-28:
@karlrees Can we add the aarch64 binary to the repo?

gnthibault, 2020-03-28:
In the mean time, I tried to build with and without libs, but it doesn't looks like it fixed the issue:

> cmake -DCMAKE_INSTALL_PREFIX=/usr/local/ -DCMAKE_BUILD_TYPE=Release -DBUILD_LIBS=0 ~/projects/indi-3rdparty/

> cmake -DCMAKE_INSTALL_PREFIX=/usr/local/ -DCMAKE_BUILD_TYPE=Release -DBUILD_LIBS=1 ~/projects/indi-3rdparty/

How can I build, just a subset of the targets ?

karlrees, 2020-03-28:
@knro Unfortunately, Ricoh does not provide an aarch64 binary.  Shoot.  I didn't think about that.  I should be able to make the indi-pentax driver functional without the Ricoh SDK, just missing a few features.  So I guess I need to figure out how to make indi-pentax compile on aarch64 without libricohcamerasdk.

karlrees, 2020-03-30:
Just submitted a pull request to take care of this.

closed 2020-04-29 by knro