indilib/indi-3rdparty #81: aag_cloudwatcher_ng driver unloads upon connect.

gnthibault, 2020-03-30:
**Describe the bug**
I tried to connect kstars indi client to a remote server that hosts aag_cloudwatcher_ng.
It works perfectly until I try to click "connect"
It looks like the corresponding driver is just "unloaded". All related windows/buttons disappear completely from the client.

**To Reproduce**
Run server with driver:
INDISKEL=/home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml indiserver -vvv /usr/local/bin/indi_aagcloudwatcher_ng
Then connect with indi client through kstars. Click the connect button.
Here is the skeleton (rename fro .txt to .xml):

[indi_aagcloudwatcher_ng_sk.txt](https://github.com/indilib/indi-3rdparty/files/4402355/indi_aagcloudwatcher_ng_sk.txt)

**Expected behavior**
I managed to connect to the aag_cloudwatcher with the regular driver (non -ng), so I would expect the ng to connect properly.

**Desktop (please complete the following information):**
Linux localhost.localdomain 4.4.180 #1 SMP Fri May 31 17:07:08 EDT 2019 aarch64 GNU/Linux
Distributor ID:	Debian
Description:	Debian GNU/Linux 9.9 (stretch)
Release:	9.9
Codename:	stretch

Code version:
commit cab70bff57659fd226d878cd2ddf727e995d94f2
Date:   Fri Feb 7 17:25:41 2020 +0300

**Log Files**
[log.txt](https://github.com/indilib/indi-3rdparty/files/4402321/log.txt)

knro, 2020-03-30:
@pardo-bsso can you please check?

pardo-bsso, 2020-04-13:
@gnthibault 
This is already fixed, you need to check master of both indi-3rdparty and indi .
Here I don't have those crashes anymore but confirmation from another user would be nice.

pardo-bsso, 2020-04-28:
@gnthibault are you okay?

@knro I can give you remote access to a device if you want to validate that this is not happening anymore to close this issue and #83

gnthibault, 2020-04-28:
@pardo-bsso sorry I didn't realized you answered, I am going to check right now with latest master commit

pardo-bsso, 2020-04-28:
No problem @gnthibault , thanks!

gnthibault, 2020-04-28:
It looks like I am experiencing the same problem, it is not clear however what is going wrong from the logs:

rock@localhost:~/projects/RemoteObservatory$ INDISKEL=/home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml indiserver -vvv /usr/local/bin/indi_aagcloudwatcher_ng
2020-04-28T16:30:53: startup: indiserver -vvv /usr/local/bin/indi_aagcloudwatcher_ng 
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: pid=7420 rfd=3 wfd=6 efd=7
2020-04-28T16:30:53: listening to port 7624 on fd 4
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: sending msg copy 1 nq 1:
<getProperties version='1.7'/>

2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: Initializing from AAG Cloud Watcher device...
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read getProperties GPS Simulator GEOGRAPHIC_COORD 
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: snooping on GPS Simulator.GEOGRAPHIC_COORD
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: Using INDISKEL /home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defSwitchVector AAG Cloud Watcher NG CONNECTION Idle rw
    CONNECT='Off'
 DISCONNECT='On'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defTextVector AAG Cloud Watcher NG DRIVER_INFO Idle ro
 DRIVER_NAME='AAG Cloud Watcher NG'
 DRIVER_EXEC='indi_aagcloudwatcher_ng'
 DRIVER_VERSION='1.5'
 DRIVER_INTERFACE='128'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defNumberVector AAG Cloud Watcher NG skyCorrection Idle rw
         k1='33'
         k2='0'
         k3='4'
         k4='100'
         k5='100'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defSwitchVector AAG Cloud Watcher NG anemometerType Idle rw
       GRAY='Off'
      BLACK='On'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defNumberVector AAG Cloud Watcher NG sensors Idle ro
 infraredSky='0'
 correctedInfraredSky='0'
 infraredSensor='0'
 rainSensor='0'
 rainSensorTemperature='0'
 rainSensorHeater='0'
 brightnessSensor='0'
 ambientTemperatureSensor='0'
  windSpeed='0'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defSwitchVector AAG Cloud Watcher NG deviceSwitch Idle rw
       open='Off'
      close='On'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defNumberVector AAG Cloud Watcher NG unitErrors Idle ro
 internalErrors='0'
 firstAddressByteErrors='0'
 commandByteErrors='0'
 secondAddressByteErrors='0'
 pecByteErrors='0'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defNumberVector AAG Cloud Watcher NG heaterParameters Idle rw
    tempLow='0'
   tempHigh='20'
   deltaLow='6'
  deltaHigh='4'
        min='10'
 heatImpulseTemp='10'
 heatImpulseDuration='60'
 heatImpulseCycle='600'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defSwitchVector AAG Cloud Watcher NG heaterStatus Idle ro
     normal='On'
 increasing='Off'
      pulse='Off'
2020-04-28T16:30:53: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read defNumberVector AAG Cloud Watcher NG readings Idle ro
     supply='0'
        sky='0'
     sensor='0'
    ambient='0'
       rain='0'
 rainHeater='0'
   rainTemp='0'
[... 1025 lines cut ...]
    </defSwitch>
    <defSwitch name="INDI_DISABLED" label="Disabled">
Off
    </defSwitch>
</defSwitchVector>

2020-04-28T16:31:32: Client 0: sending msg copy 1 nq 3:
<defSwitchVector device="AAG Cloud Watcher NG" name="DEVICE_PORT_SCAN" label="Refresh" group="Connection" state="Idle" perm="rw" rule="OneOfMany" timeout="60" timestamp="2020-04-28T16:31:32.487295602">
    <defSwitch name="Scan Ports" label="Scan Ports">
Off
    </defSwitch>
</defSwitchVector>

2020-04-28T16:31:32: Client 0: sending msg copy 1 nq 2:
<defSwitchVector device="AAG Cloud Watcher NG" name="SYSTEM_PORTS" label="System Ports" group="Connection" state="Idle" perm="rw" rule="AtMostOne" timeout="60" timestamp="2020-04-28T16:31:32.487790848">
    <defSwitch name="/dev/ttyUSB0" label="/dev/ttyUSB0">
Off
    </defSwitch>
    <defSwitch name="/dev/ttyUSB1" label="/dev/ttyUSB1">
Off
    </defSwitch>
</defSwitchVector>

2020-04-28T16:31:32: Client 0: sending msg copy 1 nq 1:
<message device="AAG Cloud Watcher NG" timestamp="2020-04-28T16:31:32.487878930" message="[INFO] No previous configuration found. To save driver configuration, click Save Configuration in Options tab"/>

gnthibault, 2020-04-28:
Biggest issue is probably here in recent set of logs:

> 2020-04-28T16:59:34: Client 0: read newSwitchVector AAG Cloud Watcher NG CONNECTION 
>     CONNECT='On'
> 2020-04-28T16:59:34: Driver /usr/local/bin/indi_aagcloudwatcher_ng: queuing responsible for <newSwitchVector device='AAG Cloud Watcher NG' name='CONNECTION'>
> 2020-04-28T16:59:34: Driver /usr/local/bin/indi_aagcloudwatcher_ng: sending msg copy 1 nq 1:
> <newSwitchVector device="AAG Cloud Watcher NG" name="CONNECTION">
>     <oneSwitch name="CONNECT">
> On
>     </oneSwitch>
> </newSwitchVector>
> 
> 2020-04-28T16:59:34: Driver /usr/local/bin/indi_aagcloudwatcher_ng: read message AAG Cloud Watcher NG   '[INFO] Connected to AAG Cloud Watcher'
> 2020-04-28T16:59:34: Client 0: queuing <message device='AAG Cloud Watcher NG' name=''>
> 2020-04-28T16:59:34: Client 0: sending msg copy 1 nq 1:
> <message device="AAG Cloud Watcher NG" timestamp="2020-04-28T16:59:34.212006458" message="[INFO] Connected to AAG Cloud Watcher"/>
> 
> Child process 8516 died
> 2020-04-28T16:59:34: Driver /usr/local/bin/indi_aagcloudwatcher_ng: stderr EOF
>

gnthibault, 2020-04-28:
Probably a good old segfault, from strace:

> write(2, "2020-04-28T17:02:03: Driver /usr"..., 1142020-04-28T17:02:03: Driver /usr/local/bin/indi_aagcloudwatcher_ng: Initializing from AAG Cloud Watcher device...
> ) = 114
> _newselect(8, [3 4 7], [], NULL, NULL)  = 2 (in [3 7])
> read(7, "Using INDISKEL /home/rock/projec"..., 49152) = 112
> gettimeofday({tv_sec=1588093323, tv_usec=3880}, NULL) = 0
> write(2, "2020-04-28T17:02:03: Driver /usr"..., 1802020-04-28T17:02:03: Driver /usr/local/bin/indi_aagcloudwatcher_ng: Using INDISKEL /home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml
> ) = 180
> read(3, "<?xml version='1.0'?>\n<getProper"..., 49152) = 100
> _newselect(8, [3 4 7], [], NULL, NULL)  = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<defSwitch"..., 49152) = 2329
> _newselect(8, [3 4 7], [], NULL, NULL)  = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<defNumber"..., 49152) = 10115
> _newselect(8, [3 4 7], [], NULL, NULL)  = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<defSwitch"..., 49152) = 3138
> _newselect(8, [3 4 7], [], NULL, NULL)  = 1 (in [4])
> accept(4, {sa_family=AF_INET, sin_port=htons(37838), sin_addr=inet_addr("192.168.0.32")}, [16]) = 0
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "<getProperties version='1.7'/>\n", 49152) = 31
> _newselect(8, [0 3 4 7], [6], NULL, NULL) = 1 (out [6])
> write(6, "<getProperties version=\"1.7\"/>\n", 31) = 31
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<defSwitch"..., 49152) = 422
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 361) = 361
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<defTextVe"..., 49152) = 591
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 2 (in [3], out [0])
> write(0, "<defTextVector device=\"AAG Cloud"..., 512) = 512
> read(3, "<?xml version='1.0'?>\n<defNumber"..., 49152) = 8435
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 2 (in [3], out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 726) = 726
> read(3, "<?xml version='1.0'?>\n<defNumber"..., 49152) = 5903
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 360) = 360
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 1392) = 1392
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 344) = 344
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 861) = 861
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 1176) = 1176
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 435) = 435
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 1458) = 1458
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defTextVector device=\"AAG Cloud"..., 258) = 258
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defNumberVector device=\"AAG Clo"..., 1687) = 1687
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 337) = 337
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 505) = 505
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 380) = 380
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defTextVector device=\"AAG Cloud"..., 260) = 260
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 806) = 806
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 374) = 374
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 295) = 295
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<defSwitchVector device=\"AAG Clo"..., 378) = 378
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "<enableBLOB device='AAG Cloud Wa"..., 49152) = 61
[... 8 lines cut ...]
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<setTextVector device=\"AAG Cloud"..., 204) = 204
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<setSwitchVector device=\"AAG Clo"..., 266) = 266
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "<newTextVector\n", 49152)      = 15
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "  device='AAG Cloud Watcher NG'\n"..., 49152) = 131
> _newselect(8, [0 3 4 7], [6], NULL, NULL) = 1 (out [6])
> write(6, "<newTextVector device=\"AAG Cloud"..., 136) = 136
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [3])
> read(3, "<?xml version='1.0'?>\n<setTextVe"..., 49152) = 237
> _newselect(8, [0 3 4 7], [0], NULL, NULL) = 1 (out [0])
> write(0, "<setTextVector device=\"AAG Cloud"..., 202) = 202
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "<newSwitchVector\n", 49152)    = 17
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [0])
> read(0, "  device='AAG Cloud Watcher NG'\n"..., 49152) = 129
> _newselect(8, [0 3 4 7], [6], NULL, NULL) = 1 (out [6])
> write(6, "<newSwitchVector device=\"AAG Clo"..., 136) = 136
> _newselect(8, [0 3 4 7], [], NULL, NULL) = 1 (in [3])
> --- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_KILLED, si_pid=8590, si_uid=1000, si_status=SIGSEGV, si_utime=1, si_stime=0} ---
> write(2, "Child process 8590 died\n", 24Child process 8590 died
> ) = 24
>

knro, 2020-04-28:
Looks like this is using a driver installed to /usr/local/bin

Are you sure this is where the latest driver is compiled and installed to? Usually INDI in /usr/local/bin means trouble since there could be another version at /usr/bin

gnthibault, 2020-04-28:
Yes I checked that was using the right version (I have nothing related to indi in /usr/bin)

/usr/local/bin/indiserver --version
2020-04-28T18:28:34: startup: /usr/local/bin/indiserver --version 
Usage: /usr/local/bin/indiserver [options] driver [driver ...]
Purpose: server for local and remote INDI drivers
INDI Library: 1.8.6
Code v1.7.8-546-g7894be27. Protocol 1.7.

/usr/local/bin/indi_aagcloudwatcher_ng --version
Initializing from AAG Cloud Watcher device...
Usage: indi_aagcloudwatcher_ng [options]
Purpose: INDI Device driver framework.
Options:
 -v    : more verbose to stderr
Segmentation fault

I guess the segfault is a pretty bad sign

gnthibault, 2020-04-28:
With strace:

strace ./indi_aagcloudwatcher_ng --version
execve("./indi_aagcloudwatcher_ng", ["./indi_aagcloudwatcher_ng", "--version"], [/* 31 vars */]) = 0
brk(NULL)                               = 0xab492000
uname({sysname="Linux", nodename="localhost.localdomain", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/v8l/neon/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/v8l/neon/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/v8l/neon/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/v8l/neon", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/v8l/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/v8l/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/v8l/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/v8l", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/neon/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/neon/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/neon/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/neon", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/tls/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/tls", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/v8l/neon/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/v8l/neon/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/v8l/neon/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/v8l/neon", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/v8l/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/v8l/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/v8l/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/v8l", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/neon/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/neon/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/neon/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/neon", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/vfp/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat64("/usr/local/lib/vfp", 0xfff76cf8) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libindidriver.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\320\317\2\0004\0\0\0"..., 512) = 512
lseek(3, 8218276, SEEK_SET)             = 8218276
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1520) = 1520
lseek(3, 853037, SEEK_SET)              = 853037
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=8219796, ...}) = 0
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xf7273000
mmap2(NULL, 1347012, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf7105000
mprotect(0xf71d2000, 61440, PROT_NONE)  = 0
mmap2(0xf71e1000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xcc000) = 0xf71e1000
mmap2(0xf71e6000, 425412, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xf71e6000
close(3)                                = 0
open("/usr/local/lib/libindiAlignmentDriver.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0xu\0\0004\0\0\0"..., 512) = 512
lseek(3, 966832, SEEK_SET)              = 966832
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1520) = 1520
lseek(3, 81969, SEEK_SET)               = 81969
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0057-A\0\6\n\7A\10\1\t\2\n\4\22"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=968352, ...}) = 0
mmap2(NULL, 147476, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xf70e0000
mprotect(0xf70f4000, 61440, PROT_NONE)  = 0
mmap2(0xf7103000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x13000) = 0xf7103000
close(3)                                = 0
open("/usr/local/lib/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("tls/v8l/neon/vfp/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("tls/v8l/neon/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
getuid32()                              = 1000
write(2, "Usage: indi_aagcloudwatcher_ng ["..., 41Usage: indi_aagcloudwatcher_ng [options]
) = 41
write(2, "Purpose: INDI Device driver fram"..., 39Purpose: INDI Device driver framework.
) = 39
write(2, "Options:\n", 9Options:
)               = 9
write(2, " -v    : more verbose to stderr\n", 32 -v    : more verbose to stderr
) = 32
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x6e2d7261} ---
+++ killed by SIGSEGV +++
Segmentation fault

knro, 2020-04-28:
Use this:

`gdb -ex "set follow-fork-mode child" -ex run --args indiserver -v indi_aagcloudwatcher_ng indi_simulator_ccd indi_simulator_telescope
`
In ekos, create a profile, set to remote (keep it localhost). Then select the Telscope, CCD, and AAG CloudWatcher NG drivers and connect. When it crashes in the console, go there and run **backtrace**

pardo-bsso, 2020-04-28:
This is really odd.
Are you positive sure that you are running with a latest build from master of both indi-3rdparty and indilib?

gnthibault, 2020-04-28:
INDISKEL=/home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml gdb -ex "set follow-fork-mode child" -ex run --args indiserver -v indi_aagcloudwatcher_ng indi_simulator_ccd indi_simulator_telescope
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from indiserver...done.
Starting program: /usr/local/bin/indiserver -v indi_aagcloudwatcher_ng indi_simulator_ccd indi_simulator_telescope
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2020-04-28T20:18:23: startup: /usr/local/bin/indiserver -v indi_aagcloudwatcher_ng indi_simulator_ccd indi_simulator_telescope 
[New process 29365]
2020-04-28T20:18:23: Driver indi_aagcloudwatcher_ng: pid=29365 rfd=3 wfd=6 efd=7
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2020-04-28T20:18:23: Driver indi_simulator_ccd: pid=29366 rfd=4 wfd=9 efd=10
2020-04-28T20:18:23: Driver indi_simulator_telescope: pid=29367 rfd=5 wfd=12 efd=13
2020-04-28T20:18:23: listening to port 7624 on fd 8
process 29365 is executing new program: /usr/local/bin/indi_aagcloudwatcher_ng
2020-04-28T20:18:23: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2020-04-28T20:18:23: Driver indi_simulator_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD
2020-04-28T20:18:23: Driver indi_simulator_telescope: snooping on GPS Simulator.TIME_UTC
2020-04-28T20:18:23: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK
2020-04-28T20:18:23: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_SHUTTER
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_SLOT
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on CCD Simulator.FILTER_NAME
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on SQM.SKY_QUALITY
2020-04-28T20:18:23: Driver indi_simulator_ccd: snooping on Focuser Simulator.FWHM
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
2020-04-28T20:18:24: Driver indi_aagcloudwatcher_ng: Initializing from AAG Cloud Watcher device...
2020-04-28T20:18:24: Driver indi_aagcloudwatcher_ng: snooping on GPS Simulator.GEOGRAPHIC_COORD
2020-04-28T20:18:24: Driver indi_aagcloudwatcher_ng: Using INDISKEL /home/rock/projects/RemoteObservatory/conf_files/indi_driver_conf/indi_aagcloudwatcher_ng_sk.xml
2020-04-28T20:18:45: Client 0: new arrival from 192.168.0.32:38302 - welcome!

Thread 2.1 "indi_aagcloudwa" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xf77e5e20 (LWP 29365)]
0xf73f40d2 in strcpy () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) backtrace
#0  0xf73f40d2 in strcpy () from /lib/arm-linux-gnueabihf/libc.so.6
#1  0xaaab01b2 in strcpy (__src=<optimized out>, __dest=0xaaacb268 "") at /usr/include/arm-linux-gnueabihf/bits/string3.h:110
#2  CloudWatcherController::getFirmwareVersion (this=this@entry=0xaaacb388, version=version@entry=0xaaacb268 "")
    at /home/rock/projects/indi-3rdparty/indi-aagcloudwatcher-ng/CloudWatcherController_ng.cpp:381
#3  0xaaab0510 in CloudWatcherController::getConstants (this=0xaaacb388, cwc=cwc@entry=0xaaacb268) at /home/rock/projects/indi-3rdparty/indi-aagcloudwatcher-ng/CloudWatcherController_ng.cpp:230
#4  0xaaaadb2c in AAGCloudWatcher::sendConstants (this=this@entry=0xaaac9078) at /home/rock/projects/indi-3rdparty/indi-aagcloudwatcher-ng/indi_aagcloudwatcher_ng.cpp:972
#5  0xaaaae76c in AAGCloudWatcher::Handshake (this=0xaaac9078) at /home/rock/projects/indi-3rdparty/indi-aagcloudwatcher-ng/indi_aagcloudwatcher_ng.cpp:70
#6  0xf76f45d6 in INDI::Weather::callHandshake (this=<optimized out>) at /home/rock/projects/indi/libs/indibase/indiweather.cpp:398
#7  0xf76f460a in INDI::Weather::<lambda()>::operator() (__closure=<optimized out>) at /home/rock/projects/indi/libs/indibase/indiweather.cpp:82
#8  std::_Function_handler<bool(), INDI::Weather::initProperties()::<lambda()> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/6/functional:1717
[... cut ...]
#19 0xf76b0eaa in callCallback (rfdp=0xfffef2cc) at /home/rock/projects/indi/eventloop.c:347
#20 oneLoop () at /home/rock/projects/indi/eventloop.c:439
#21 0xf76b0f20 in eventLoop () at /home/rock/projects/indi/eventloop.c:106
#22 0xf76b0bae in main (ac=0, av=<optimized out>) at /home/rock/projects/indi/indidrivermain.c:98
#23 0xf73b44aa in __libc_start_main () from /lib/arm-linux-gnueabihf/libc.so.6
#24 0xaaaad22c in _start ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

pardo-bsso, 2020-04-28:
This is strange.

What's the firmware version reported when you use the old driver?
Could you try using an x86 platform?
Would you be willing to grant remote access to the device using something like a socat tunnel?

gnthibault, 2020-04-29:
Firmware version as reported by the non -ng driver: 5.54
Internal Serial: 1389
Driver version: indi_aagcloudwatcher 1.4

pardo-bsso, 2020-04-30:
Hi all.

@gnthibault I tried a fresh clone of both indi and indi-3rdparty under QEMU using an aarch64 version of Debian Stretch (mostly following the steps at https://wiki.debian.org/Arm64Qemu) and I can't reproduce the crash using the commands you show.

The only thing I do differently is using a tcp connection to the device as I do not have physical access to it for the moment.

How are you configuring and building indi and indi-3rdparty from source?
Do you have something special set on the environment variables PATH, LD_LIBRARY_PATH and LD_PRELOAD?
Do you see something on the output of dmesg after the crash?

Thanks.

gnthibault, 2020-04-30:
Hmm there might be something that cause trouble. My os kernel is aarch64 but all userspace is arm 32. I might actually need to force that in cmake cache file. Do you think it can ve the source of the problem ?

pardo-bsso, 2020-04-30:
That seems odd. If possible I'd try with a fresh git clone and empty cache.

What motivated you to use a 32 bit userspace? Is there any guide I can use to set up a system so it is like yours?

On the other hand:

```
write(2, " -v : more verbose to stderr\n", 32 -v : more verbose to stderr
) = 32
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x6e2d7261} ---
```

But the length of  " -v : more verbose to stderr\n" is 29 and write() is called with a length of 32.

And also:

```
Thread 2.1 "indi_aagcloudwa" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0xf77e5e20 (LWP 29365)]
0xf73f40d2 in strcpy () from /lib/arm-linux-gnueabihf/libc.so.6
(gdb) backtrace
#0 0xf73f40d2 in strcpy () from /lib/arm-linux-gnueabihf/libc.so.6
#1 0xaaab01b2 in strcpy (__src=, __dest=0xaaacb268 "") at /usr/include/arm-linux-gnueabihf/bits/string3.h:110
```

gnthibault, 2020-05-02:
Ok, I am about to test this on a much more integrated setup, on a raspberry pi 4, that runs the wonderful indiwebmanager from @knro .

Do you know what is the golden standard to run aag_cloudwatcher_ng from indiwebmanager ?

gnthibault, 2020-05-02:
OK I tested everything on a raspberry pi 4 with ubuntu 20.04 aarch64 and everything is working like a charm, I'll close the PR

closed 2020-05-02 by gnthibault