indilib/indi-3rdparty #83: AAGCloudWatcher NG Crash

knro, 2020-04-01:
In [isWetRain](https://github.com/indilib/indi-3rdparty/blob/263ae0342828a1380248f5b9392160749531d236/indi-aagcloudwatcher-ng/indi_aagcloudwatcher_ng.cpp#L393) function, it checks for **rainConditions** switch, but I don't see this switch getting defined anywhere? Right now it crashes, I added a condition to check if the pointer is null or not, but the property needs to be defined somewhere.

@pardo-bsso Do you know what the issue is?

pardo-bsso, 2020-04-01:
Ok my bad. This was left over from when I moved the code over to 3rdparty.

What I need to do to retain the behaviour of the previous driver is to check is if the parameter WEATHER_RAIN is within safe limits.

@gnthibault this is also the cause of what you see at #81

pardo-bsso, 2020-04-01:
@knro 

I'd probably tackle this by extracting part of [WeatherInterface::syncCriticalParameters()](https://github.com/indilib/indi/blob/master/libs/indibase/indiweatherinterface.cpp#L190) so that subclasses can check individually if a parameter is in the safe zone.

Any thoughts?

knro, 2020-04-02:
Ok, so the check in syncCriticalParameters is not sufficient enough? At any rate, please take whatever necessary measures to resolve this issue.

knro, 2020-04-29:
Duplicate of #81

closed 2020-04-29 by knro