indilib/indi-3rdparty #87: Problem compiling libricohcamerasdk

rkaczorek, 2020-04-06:
Apparently there's a problem in packaging libricohcamerasdk on raspberry pi (raspbian buster).
Compilation seems ok, but debian packaging returns error:
`dh_strip -plibricohcamerasdk  
objcopy: debian/.debhelper/libricohcamerasdk/dbgsym-root/usr/lib/debug/.build-id/f0/74b2974b6314837f82e0da73d645ef028a7564.debug: not enough room for program headers, try linking with -N
objcopy:debian/.debhelper/libricohcamerasdk/dbgsym-root/usr/lib/debug/.build-id/f0/74b2974b6314837f82e0da73d645ef028a7564.debug[.note.gnu.build-id]: bad value
dh_strip: objcopy --only-keep-debug --compress-debug-sections debian/libricohcamerasdk/usr/lib/arm-linux-gnueabihf/libRicohCameraSDKCpp.so.1.1.0 debian/.debhelper/libricohcamerasdk/dbgsym-root/usr/lib/debug/.build-id/f0/74b2974b6314837f82e0da73d645ef028a7564.debug returned exit code 1
dh_strip: Aborting due to earlier error
make: *** [/usr/share/cdbs/1/rules/debhelper.mk:295: binary-strip-IMPL/libricohcamerasdk] Error 2
`
@karlrees any ideas?

karlrees, 2020-04-07:
My guess, just from a quick search on the error, is that this is the issue: https://bugs.debian.org/cgi-bin/%3Ca%20href=%22bugreport.cgi?bug=903396%22%3E903396%3C/a%3E

I'll need to go back and check which version of patchelf I used to fix the library file.  Might be able to solve this by redoing it with an upgraded version.

Unfortunately, I don't know how to test the packaging to see if what I do works (pointers would be appreciated).

knro, 2020-04-07:
You can test the Debian packages directly by copying the debian directory of the package to the source directory of what you want to build (e.g. libricohcamerasdk) then in that directly run this:

`debuild -us -uc`

Then it should build the corresponding .deb file. Then to verify install the deb file and make sure all is OK.

karlrees, 2020-04-07:
I must have something configured incorrectly.  At first I was getting errors about no upstream tarball being found.  Then I changed the source format from native to quilt, and got further, but it was expecting a libricohcohcamersdk/libricohcamerasdk source directory.  So then I changed the deb source directory to ., and got further, but now I'm stuck at a no makefile being found error in obj-arm-linux-gnueabihf...

knro, 2020-04-08:
I found couple of issues in the debian files and pushed a fix. So here is the workflow I did:

```
mkdir packaging && cd packaging
cp -rf ~/Projects/indi-3rdparty/libricohcamerasdk .
cp -rf ~/Projects/indi-3rdparty/cmake_modules libricohcamerasdk
tar -czf libricohcamerasdk_1.1.0.orig.tar.gz libricohcamerasdk
cp -rf ~/Projects/indi-3rdparty/debian/libricohcamerasdk libricohcamerasdk/debian
cd libricohcamerasdk
debuild -us -uc
```

Then it built the .deb file

rkaczorek, 2020-04-18:
It looks like it has been fixed. Closing

closed 2020-04-18 by rkaczorek