indilib/indi-3rdparty #90: Pentax K3II exposure time

nados970, 2020-04-10:
I have compiled the driver for pentax. I have a K3II, and I cannot change the exposure time.
Each photo is taken with an exposure time of 1/17 even if I put 1 minute or 30 seconds.
With the indi_pentax_ccd driver I have the same problem.

nados970, 2020-04-11:
In bulb mode type 2, the triggering is done but I have to press again to stop the photo.
In bulb mode type 1 the exposure time of the photo is 1/17s. 
I have the same result with the driver indi_pentax and indi_pentax_ccd.
I'm in MSC mode.

knro, 2020-04-12:
@karlrees Any ideas?

nados970, 2020-04-12:
Maybe we should do a bulb_T2 function with: 
trigger, 
sleep (time variable), 
trigger

nados970, 2020-04-12:
I see in the pktriggercord code two functions, bulb_old and bulb_new.  
Which mode does the K3II use?

karlrees, 2020-04-12:
Can you let me know what results you get when you use pktriggercord (gui or cli)?  I don't have a K3II to test, but I think the issue is related to this pktriggercord issue: https://github.com/asalamon74/pktriggercord/issues/25.  

Can you set the exposure time in non-bulb modes?  

Also, I wouldn't expect it to work for the K3II, but out of curiosity did you test PTP mode?  Thanks.

nados970, 2020-04-12:
in ptp mode, I cannot trigger. 
I tried with Kstars / Ekos then with pktriggercord-gui.  
I tried to activate Astrotracer but without result.  With Astrotracer, I can't trigger.

nados970, 2020-04-12:
With pktriggercord (gui), in M or Green mode (I have not tried the others), the exposure time works

karlrees, 2020-04-13:
So in MSC mode, indi_pentax is just wrapping the pktriggercord code.  I *think* it should be able to do anything the pktriggercord GUI can.  Can you verify whether you tried M mode with the indi_pentax driver?  Also, can you try Bulb mode with the pktriggercord GUI?

nados970, 2020-04-13:
M mode works with indi_pentax but if I put 30sec, the photo has 6s exposure.  If I put 20s, the photo has 30s of exposure.
in mode b type 1, with pktriggercord-gui, 1 / 17s.  in type 2, I have to physically press the trigger.

nados970, 2020-04-13:
With pktriggercord-gui, exposure time is correct in M ​​mode.

nados970, 2020-04-13:
other information, I'm in U3 (user mode) which is set to B mode

karlrees, 2020-04-14:
I'm not sure why you would be getting different results with the indi_pentax driver than what you get in the pktriggermode GUI.  I looked through the cli code, which is what I'm wrapping, and there was one small difference in the way I was setting shutter speed.  I updated my code to do it the same way.  You could clone my fork and try it if you to see if it makes a difference.

As for bulb mode, I don't see anywhere in the GUI code where it actually differentiates between type 1 or type 2 bulb mode.  Are you just changing the mode on the camera itself?  In type 2, are you physically pressing the trigger on the camera?  For comparison purposes, on the K-70 (which is the only system I have to test), it makes no difference between type 1 and type 2.  I just click "take picture" and it works, but perhaps that is because the K-70 uses bulb_new instead of bulb_old?

nados970, 2020-04-14:
I looked at the code and actually I didn't see any difference in the code between Type 1 and Type 2. That's why I wonder if it should not be a shutterT2 function with type 2, ie triggering, sleep, triggering, since the trigger works. 
I switch from type 1 to type 2 by going into the menu of the K3II and then I connect in USB

karlrees, 2020-04-19:
Were you by chance able to test whether the change I made in my fork fixed the shutter speed problem you were having while in M mode?

For bulb mode, just to verify, there's no non-manual way you can get this to work using the pktriggercord gui or cli right now?  If so, can you log the issue with pktriggercord?  If they can fix it, I can just update the code here.  They have a much better idea of what's going on with the underlying code than I do, and perhaps even the right camera to test with.

nados970, 2020-04-24:
Hello,
Sorry for the late reply. I just tried with your repository and there is change. 
In B type 1 mode, if I take 3 seconds, it takes one photo at the beginning (1/17s) and another after 3 seconds (1/17s also). 
In Type 2, no change.

nados970, 2020-04-24:
In M mode if I take 3 seconds, the photo is well taken with a 3-second pose. it works but I'm limited to 30 seconds, which is better than nothing. On the other hand if I take 3 photos of 30 seconds, the first one makes 30 secconds of poses, the second I see the transfer after 3 seconds while the trigger is still open and the third is 30 seconds.

karlrees, 2020-04-27:
I tried out a number of scenarios to see if I could replicate the sequence problem you're seeing in M mode.  In most cases, it works fine for me.  But, occasionally if I switch between M and B mode without restarting the camera and indiserver, it seems like I get some captures that start and don't finish, and then when the next capture starts it sees something to download and grabs that instead of waiting for the new exposure.  It seems a little random to me right now, but any chance that's the same thing you're seeing?  Stated another way, if you start out in M mode and don't change anything, do you still have the sequence problem?

Also, do you have any error logs?  What equipment are you running on?

nados970, 2020-06-14:
With indi_gphoto_ccd and shutter release (cable diy), it's work.

karlrees, 2020-06-15:
Yeah, it'd be nice if this would work without the diy cable, though.  Someone else seems to be having similar problems with the K-30 and just sent me some logs.  Gonna take a closer look at it when I get some time to see if I can get to the bottom of it.

nados970, 2021-04-27:
[OneShootK3II-Flucard.zip](https://github.com/indilib/indi-3rdparty/files/6383177/OneShootK3II-Flucard.zip)

here's a wireshark capture made with the K3II and a flucard, if that helps.

knro, 2021-11-24:
@karlrees Any update on this issue?

karlrees, 2021-11-25:
No.  I think it's just a limitation of the pktriggercord-based driver.  This is an interesting thread though: https://github.com/asalamon74/pktriggercord/issues/51.  I wonder if the solution at the end would help.

jesshurch, 2022-01-14:
Just a note to say I'm seeing similar on a K3(1) both running firmware 1.41 and 1.43 both via indi and directly in pktriggercord.
In bulb mode I only get a single 1/17 sec exposure rather than the two that some others mention elsewhere