indilib/indi-3rdparty #104: indi_gphoto - Canon DSLR never returns to Bulb mode after sub 30s exposures

Phobos2350, 2020-04-23:
**Description**
_Module_
indi_gphoto

_Scenario_
Using a Canon 400D with no dedicated bulb widget/over USB bulb shutter control relies on camera set to 'Bulb' exposure manually on-camera and a shutter release cable.
Before the change [df46e97 on 2020-03-23](https://github.com/indilib/indi-3rdparty/commit/df46e97180106e3c114db8f8d8176bc41dd8acfb) all exposures when 'Force Bulb' is switched ON resulted in all exposures taken via the shutter release commands, whether the camera supports them or not.

Since updating now the module reverts to finding the best matching on-camera exposure setting and using that as the exposure setting, ignoring 'Force Bulb ON' in cases under the RELEASE_SHUTTER_THRESHOLD of 30 seconds.

When taking an exposure below 30 seconds the camera behaves well and expected, but changing to an exposure >30s the camera never reverts to 'Bulb' exposure mode on-camera and remains at the last set exposure time <30s that was taken. This is only solved by manually setting the camera by hand back into Bulb exposure time using the physical exposure wheel/dial. 

However the camera remains locked in an exposure the entire length of the requested bulb shutter time, and the shutter release remains active even after the camera stops the exposure and the physical shutter is heard closing. The final file is then returned as an _x_ length exposure but is in reality an exposure of length _y_ where _y_ is the last set <30s exposure value that was last taken and corresponds to the on-camera exposure setting that is displayed on the camera display.

**To Reproduce**
1. Ensure the DSLR is in Manual Mode, with Bulb exposure length set on-camera
2. Take an exposure of length _x_ above 30s using indi_gphoto through Ekos
3. Receive image as expected of length _x_
4. Take and exposure of length _y_ below 30s
5. Receive image as expected of length _y_
6. Set exposure of length _x_ above 30s again
7. Camera stops exposing after time _y_, but remains locked in an exposure of length _x_ and returns an image with all associated meta data/header data as if it is of length _x_ but the actual image is consistent with one of length _y_ in terms of noise/data.
8. Check on-camera display and it remains set to length _y_ exposure length
9. Set camera back to 'Bulb' exposure length manually
10. Process repeats

**Expected behavior**
In scenarios where an exposure above 30s is taken, the camera should be returned to Bulb exposure length/mode even if it has been set below 30s by a previous exposure.

However the Canon 400D only reports back to the exposure widget that it is capable of up to 30s exposures and never lists a Bulb mode as is present in the _fallbackShutterSpeeds[]_ array, so this mode is never chosen again once it leaves it for another supported exposure length and there seems to be no way of putting the 400D back into Bulb exposure length once it has been switched out of it. The only way of restoring >30s Bulb exposure behaviour is to manually reset the camera to Bulb mode through the physical camera controls.

**Suggested Fix**
Modify the change made to Bulb shutter mode in commit [df46e97 on 2020-03-23](https://github.com/indilib/indi-3rdparty/commit/df46e97180106e3c114db8f8d8176bc41dd8acfb) and ensure 'Force Bulb' remains strict and only uses the shutter release method of exposure length control rather than falling back to on-camera exposure length below 30s to ensure cameras like the 400D without Bulb shutter support (without a shutter release) can remain forever in the manually set Bulb mode on-camera without manual intervention, regardless of requested exposure times <30s.

Changing line 1071 of gphoto_driver.cpp from OR to AND ensures that 'Force Bulb' remains strictly enabled regardless of whether the camera supports the requested sub 30s exposure length
FROM
`if (gphoto->force_bulb == false ||
            ((gphoto->bulb_port[0] || gphoto->dsusb) && exptime_usec <= RELEASE_SHUTTER_THRESHOLD))`

TO
`if (gphoto->force_bulb == false &&
            ((gphoto->bulb_port[0] || gphoto->dsusb) && exptime_usec <= RELEASE_SHUTTER_THRESHOLD))`

This change will make 'Force Bulb' setting consistent over all exposure lengths and never fall back to on-camera exposure length settings, essentially nullifying the expected behaviour of 'Force Bulb' as a setting.

This change has been compiled and tested on a range of exposure times from 1/4000 (0.00025s) to 300s on the Canon 400D only and returns behaviour to as it was before the commit which caused these issues. Changing exposure times below and above the 30s threshold does not change behavious and Bulb Mode remains consistent and active on-camera.

**Desktop**
 - OS: Raspbian 10 Buster
 - Version: indilib-3rdparty v1.8.5
 - Build: 2020-04-13T21:01:23Z
 - Environment: KStars and Ekos 

**Log Files**
With the described behaviour returning exposures without reverting to bulb mode.
Note it can be seen when the exposure widget is queried for exposure times no 'BULB' length is returning in index 0.
[log_15-31-03.zip](https://github.com/indilib/indi-3rdparty/files/4524573/log_15-31-03.zip)

With the single || to && change as described which produces expsosures of length 0.00025 - 300 seconds tested
[log_17-37-10.zip](https://github.com/indilib/indi-3rdparty/files/4524580/log_17-37-10.zip)

knro, 2020-04-30:
Actually, now that I think about it. If exposure widget is always set to BULB before > 30 seconds bulb exposure it taken, wouldn't that solve the issue as well?

Phobos2350, 2020-04-30:
I can confirm it is now picking up the 'bulb' string when parsing shutter speeds from the camera on initalisation which it wasn't previously.

`[2020-04-30T14:54:33.853 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Parsing shutter speed #52: bulb "`

`[2020-04-30T14:54:33.853 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] exposure[52]= BULB "`

Attempting a 5 second capture with Force Bulb enabled:

`[2020-04-30T14:58:46.886 BST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 5.000-second  images, has 0/1 frames captured and will be processed."`
`[2020-04-30T14:58:46.904 BST DEBG ][           org.kde.kstars.indi] - Requested exposure 5 closes match is 5`
`[2020-04-30T14:58:46.905 BST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 5.000-second  image..."`
`[2020-04-30T14:58:46.952 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Starting 5 seconds exposure. "`

Works for 30 seconds as well. On-camera display remains at 'Bulb' setting for any exposure length as expected with Force Bulb enabled.

Now a 35 second capture with Force Bulb:

`[2020-04-30T15:01:45.180 BST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 35.000-second  images, has 0/1 frames captured and will be processed."`
`[2020-04-30T15:01:45.198 BST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 35.000-second  image..."`
`[2020-04-30T15:01:45.275 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Starting 35 seconds exposure. "`

Confirmed by timing the camera manaually that exposure lasts 35s and the on-camera display remains at 'Bulb' setting. I then repeated the test for 5s and 30s exposures and it behaves the same which is good.

So Force Bulb: ON seems to be working in this case and the camera is only using the shutter release cable and bulb exposure mode for all captures.

Now I set Force Bulb to OFF and try a 5s exposure:

`[2020-04-30T15:05:36.827 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Force BULB is disabled. Exposures shall utilize camera predefined exposures time first before attempting BULB. "`

`[2020-04-30T15:05:45.738 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Starting 5 seconds exposure. "`
`[2020-04-30T15:05:45.766 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Finding optimal exposure setting for 5 seconds in shutterspeed (count=53)... "`
`[2020-04-30T15:05:45.767 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Closest match: 5 seconds Index: 8 "`
`[2020-04-30T15:05:45.767 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed: 8 (5) "`
`[2020-04-30T15:05:45.927 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T15:05:45.927 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Using predefined exposure time: 5 seconds "`

This works as expected, on-camera display shows a 5" exposure is selected through the camera exposure widget over USB.

I repeat the test for 30s and the same results are observed.

Now I try a 35s exposure with Force Bulb OFF:

`[2020-04-30T15:07:12.508 BST INFO ][   org.kde.kstars.ekos.capture] - "Job requires 35.000-second  images, has 0/1 frames captured and will be processed."`
`[2020-04-30T15:07:12.525 BST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 35.000-second  image..."`
`[2020-04-30T15:07:12.549 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Starting 35 seconds exposure. "`
`[2020-04-30T15:07:12.617 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Opening remote serial shutter port: /dev/shutter ... "`

Timing the camera manually the camera only exposes for 30s and the on-camera display reads 30" and not 'Bulb' as expected. The exposure never gets set back to 'Bulb' by the exposure controller/widget.

I repeat the test for 5s and the camera is set for a 5" on-camera exposure and takes a 5s image which I timed.

I then attempt a 60s exposure with Force Bulb still OFF and get a repeat of the 35s behaviour, this time the camera only exposes for the previously set 5s exposure time and on-camera display reads 5" not 'Bulb'. Indi report a full 60s exposure however and continues holding the remote shutter open for the full 60 seconds even after the camera is heard to move the mirror back down and stop the exposure. The file is mis-labelled as a 60s exposure even though it is physically a 5s exposure.

I repeated the test again for 45s exposure and again only a 5s exposure is taken and the same behaviour is observed from INDI, reporting it as a full 45s exposure.

I then tried another 5s exposure while keeping Force Bulb OFF after the two failed exposures:

`[2020-04-30T15:11:17.275 BST DEBG ][           org.kde.kstars.indi] - Requested exposure 5 closes match is 5`
`[2020-04-30T15:11:17.320 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Starting 5 seconds exposure. "`
`[2020-04-30T15:11:17.358 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Finding optimal exposure setting for 5 seconds in shutterspeed (count=53)... "`
`[2020-04-30T15:11:17.359 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Closest match: 5 seconds Index: 8 "`
`[2020-04-30T15:11:17.359 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed: 8 (5) "`
[... cut ...]
- Force Bulb ON is working as expected again and keeps the camera set to 'Bulb' within the on-camera settings and all exposure time is controlled with the remote shutter release cable successfully.
- Force Bulb OFF results in unstable behaviour once the camera is taken out of 'Bulb' into a pre-defined exposure time/mode and is never able to get set back into 'Bulb' exposures and eventually fails completely.

I have included the full logs again for reference.

[log_14-54-09.zip](https://github.com/indilib/indi-3rdparty/files/4559039/log_14-54-09.zip)

Phobos2350, 2020-04-30:
Log results of a test of a full range of exposures with Force Bulb: On

Behaviour is as previous and expected allowing for a full range of camera-supported and unsupported bulb length exposures all through a forced bulb shutter usage to control exposure time exclusively.

This translated to a typical imaging session and would allow normal usage without having to manaully set the camera back to Bulb mode as it remains in Bulb mode for any length of exposure.

One observed quirk is that for minimum length exposure time of 0.00025s (1/4000) the first exposure attempt fails and the camera hangs for approx 30 seconds before restarting a second exposure attempt from scratch (note the timestamps):

`[2020-04-30T16:25:23.258 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Event timed out #1, retrying... "`
`...`
`[2020-04-30T16:25:51.353 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Event timed out #10, retrying... "`
`[2020-04-30T16:25:51.357 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[ERROR] Exposure failed to save image... Unspecified error "`
`[2020-04-30T16:25:51.364 BST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #1"`
`...`
`[2020-04-30T16:25:51.549 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Exposure complete. "`

Not entirely sure what's going on here but all frames in the imaging sequence were eventually completely successfully, in this case:
- 5s x 10
- 60s x2
- 300s x2
- 0.5s x10
- 60s x2
- 0.00025s x10 (these failed on the first capture attempt for every capture, taking approx 6mins to complete the sequence of 10 frames when the expected time should be more like 20s including download times)
- 120s x 2
- 60s x1
- 10s x1
- 60s x2
- 20s x1
- 60s x 1

In conclusion from my testing so far the Force Bulb On mode works as expected for the Canon 400D aside from failing at the first attempt of every 0.00025s exposure, but eventually completing them and a whole sequence of varied length images.

Setting Force Bulb Off results in unstable behaviour where the camera is never set back to 'Bulb' exposure lengths on the on-camera display and eventually the camera/driver fails and requires a reset.

knro, 2020-04-30:
Thank you for the great report! I just pushed more changes to track the exposure widget index. Can you please git pull and repeat your test with FORCE BULB ON, then again with OFF?

Phobos2350, 2020-04-30:
Ok with latest commit:

With FORCE BULB ON for the sequence:
- 5s x1
- 10s x1
- 10s x1 changed ISO to 1600
- 30s x1 changed ISO back to 400
- 35s x1
- 5s x1
- 35s x1
- 0.5s x1
- 0.00025s x5
- 10s x1
- 300s x1

Sequence completes successfully with the same failure on the first attempt for the 0.00025s exposures which is an annoyance but still working. I added an ISO change into the mix just to make sure this was still working ok as a precaution, which it did, so not to worry!

Logs show Exposure Widget picks up the Bulb shutter speed at index 52 once again:
`[2020-04-30T21:36:19.610 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] *  52: bulb "`

Logs for the first exposure read as follows with the new debug info present:

`[2020-04-30T21:41:56.292 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Starting exposure (exptime: 5 secs, mirror lock: 0, force bulb: true, exposure index: 0) "`
...
`[2020-04-30T21:41:56.294 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget iso: 2 (400) "`
`[2020-04-30T21:41:56.294 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:41:56.294 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget imageformat: 6 (RAW) "`
`[2020-04-30T21:41:56.294 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:41:56.295 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting exposure widget bulb index: 52 "`
`[2020-04-30T21:41:56.297 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed: 52 (bulb) "`
`[2020-04-30T21:41:56.297 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:41:56.298 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Opening remote serial shutter port: /dev/shutter ... "`

Onto the second exposure the 'Setting radio/menu widget shutterspeed' does not get called again as in the first 5s exposure:

`[2020-04-30T21:42:09.463 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget iso: 2 (400) "`
`[2020-04-30T21:42:09.463 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:42:09.463 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget imageformat: 6 (RAW) "`
`[2020-04-30T21:42:09.463 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:42:09.463 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Opening remote serial shutter port: /dev/shutter ... "`

However it all works okay and the entire sequence completes succesfully as I already stated.

Moving to FORCE BULB OFF I face immediate problems again and attempting the same sequence as above starting with a 5s exposure:

`[2020-04-30T21:53:53.648 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[INFO] Force BULB is disabled. Exposures shall utilize camera predefined exposures time first before attempting BULB. "`

`[2020-04-30T21:54:26.108 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Starting exposure (exptime: 5 secs, mirror lock: 0, force bulb: false, exposure index: 0) "`
...
`[2020-04-30T21:54:26.111 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Finding optimal exposure setting for 5 seconds in shutterspeed (count=53)... "`
`[2020-04-30T21:54:26.111 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Closest match: 5 seconds Index: 8 "`
`[2020-04-30T21:54:26.111 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed: 8 (5) "`
`[2020-04-30T21:54:26.133 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Failed to set new configuration value (camera busy), retrying... "`
`[2020-04-30T21:54:26.655 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`
`[2020-04-30T21:54:26.655 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Using predefined exposure time: 5 seconds "`
`[2020-04-30T21:54:26.655 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Exposure started. "`

The exposure however never starts within the camera and it remains at 'Bulb' on the on-camera display:

`[2020-04-30T21:54:31.557 BST INFO ][   org.kde.kstars.ekos.capture] - "Capture failed. Check INDI Control Panel for details."`
`[2020-04-30T21:54:31.561 BST INFO ][   org.kde.kstars.ekos.capture] - "Restarting capture attempt #1"`
`[2020-04-30T21:54:31.584 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Reading exposure... "`
`[2020-04-30T21:54:31.584 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Exposure complete. "`
`[2020-04-30T21:54:31.585 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Downloading image... Name: () Folder: () Delete from SD card? (false) "`
`[2020-04-30T21:54:31.588 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[ERROR] Error downloading image from camera: Directory not found "`
`[2020-04-30T21:54:31.592 BST INFO ][           org.kde.kstars.indi] - GPhoto CCD :  "[ERROR] Exposure failed to save image... Directory not found "`
[... cut ...]
`[2020-04-30T21:57:32.244 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed: 8 (5) "`
`[2020-04-30T21:57:32.244 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Failed to set new configuration value (camera busy), retrying... "`
`[2020-04-30T21:57:32.737 BST DEBG ][           org.kde.kstars.indi] - GPhoto CCD : "[DEBUG] Setting new configuration OK. "`

[log_21-35-50.zip](https://github.com/indilib/indi-3rdparty/files/4561491/log_21-35-50.zip)

knro, 2020-05-08:
Alright, I'd consider this resolved for now. Please reopen if you have further issues.

commit 6170df38 (2020-04-29): When force BULB is enabled, always enforce it regardless of duration.

Fixes #104
closed 2020-05-08 by knro