indilib/indi-3rdparty #105: Atik Horizon 2 not detected

TallFurryMan, 2020-04-26:
**Describe the bug**
The Atik INDI driver does not detect the CMOS "Horizon II" camera. The default "No camera detected. Power on?" pop-up appears twice with KStars/Ekos, and the driver is not started in the INDI panel. The unexpected behavior is consistent from the command line.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect the Atik Horizon 2 to the Linux computer, in a valid USB2 port.
2. Run `indi_atik_ccd`: `indiserver -l /home/tallfurryman/dev/tmp -p 7626 -vvv indi_atik_ccd`.
3. Message appears in log: `No Atik cameras detected. Power on?`

**Expected behavior**
Atik Horizon 2 is detected and can be used by INDI clients. This same test is able to detect and operate an Atik 314E without issues.

**Screenshots**
Output of `dmesg` relevant for the port the camera is plugged in, showing EHCI:
[image]
Output of `dmesg` when connecting the camera:
[image]
Output of `lsusb` when the camera is connected (Atik Horizon 2 on USB2 port 1, ZWO EFW and Arduino on USB1.1 ports 3 and 4:
[image]

**Desktop (please complete the following information):**
 - OS: Lubuntu 18.04
 - Version INDI 3rd-party 7c25c271
 - Atik library in /usr/local/lib/libatikcameras.so.2.2.6

**Log Files**
The only content of the log generated by the test above is the following:
```
2020-04-26T12:51:42: : No Atik cameras detected. Power on?
```
The command line shows several additional lines:
[image]

TallFurryMan, 2020-04-29:
The first message returned by Atik Support is that the Horizon 2 camera is NOT supported under Linux.

I am trying to understand the situation with them before I am forced to return the camera.

Do we have an official or unofficial list of USB identifiers recognized by `libatik`? I searched astrobin for Atik and INDI and could enumerate the following: 314E, 314L+, 383L+, Titan, 414EX, 428EX, 460EX.

TallFurryMan, 2020-04-29:
The post at https://forums.atik-cameras.com/index.php?topic=5024.0 says the Horizon 1 is supported on Linux (too bad we can't register on that forum at the moment).

TallFurryMan, 2020-04-29:
Some additional information on `dmesg` when connecting the camera:
```
Apr 26 14:38:26 AstroMaster kernel: [  226.448204] usb 1-1: new high-speed USB device number 4 using ehci-pci 
Apr 26 14:38:26 AstroMaster kernel: [  226.605007] usb 1-1: New USB device found, idVendor=20e7, idProduct=dfc3
Apr 26 14:38:26 AstroMaster kernel: [  226.605021] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 26 14:38:26 AstroMaster kernel: [  226.605030] usb 1-1: Product: WestBridge 
Apr 26 14:38:26 AstroMaster kernel: [  226.605038] usb 1-1: Manufacturer: Cypress
Apr 26 14:38:26 AstroMaster kernel: [  226.605046] usb 1-1: SerialNumber: 0000000004BE
Apr 26 14:38:26 AstroMaster mtp-probe: checking bus 1, device 4: "/sys/devices/pci0000:00/0000:00:1d.7/usb1/1-1"
Apr 26 14:38:26 AstroMaster mtp-probe: bus: 1, device: 4 was not an MTP device
Apr 26 14:38:26 AstroMaster upowerd[958]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:1d.7/usb1/1-1
```
I realize the udev rules provided by `libatik` in `99-atik.rules` are NOT installed on my system (not needed for 314E). I will run additional tests with those rules installed.

TallFurryMan, 2020-04-30:
In parallel, I see that https://www.atik-cameras.com/change-log/ states DLL V142 fixes an issue with the connection to Horizon cameras. There is NO explicit relation between V142 and what could be a v1.4.2 SDK. The first reference to the Horizon 2 in the changelog is for DLL V134.

For reference again, INDI's current `indi_atik_ccd` is built against SDK v1.3.1.

TallFurryMan, 2020-04-30:
The win32 Atik SDK 1.3.1 is using whatever DLL is available on the system.

The Atik Example Application outputs:
```
Artemis HSC Demo
Version: 143
```
This refers on my up-to-date system to version v1.4.0.3 of `AtikCameras.dll`, located in `C:\Window\system32`.

Adding a call to `SetDllDirectory` before `LoadLibrary` in `AtikCameras.cpp` to target the SDK library `lib\win\64` changes the output to:
```
Artemis HSC Demo
Version: 139
```

As a side note, this shows that SDK v1.3.1 does contain DLL V139. The SDK v1.3.1 DLL is versioned `1.3.0.9`, which, hum, matches.

TallFurryMan, 2020-04-30:
I patched the INDI `indi_atik_ccd` to retry the connection, and also to output the API and DLL version of the installed Atik library (again with an Horizon 2 connected):
```
$ indiserver -l /home/tallfurryman/dev/tmp/ -p 7626 -vvv indi_atik_ccd
2020-04-30T16:33:43: startup: indiserver -l /home/tallfurryman/dev/tmp/ -p 7626 -vvv indi_atik_ccd 
2020-04-30T16:33:43: Driver indi_atik_ccd: pid=5815 rfd=3 wfd=6 efd=7
2020-04-30T16:33:43: listening to port 7626 on fd 4
2020-04-30T16:33:43: Driver indi_atik_ccd: sending msg copy 1 nq 1:
<getProperties version='1.7'/>

2020-04-30T16:33:43: Driver indi_atik_ccd: read message    'Atik Library API V118 DLL V118 initializing...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected, retrying 1/5...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected, retrying 2/5...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected, retrying 3/5...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected, retrying 4/5...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected, retrying 5/5...'
2020-04-30T16:33:44: Driver indi_atik_ccd: read message    'No Atik Cameras detected. Power on?'
``` 

Obviously `V118` was quite unexpected!! Now checking the build procedure for `libatik`, and which 3rd party library it is using.

TallFurryMan, 2020-05-01:
To be very sure, I verified that my previous test on `Lubuntu 18.04 x86` with `indi_atik_ccd` was really using the library from the SDK v1.3.1. MD5 sums are identical.

TallFurryMan, 2020-05-01:
The issue is the same on `StellarMateOS 1.3 armv7l`, `indi_atik_ccd` does not detect the Horizon 2. I won't investigate much more as this seems only the Windows DLL were updated in this SDK.

In the meantime I received news from Atik Support acknowledging their SDK may be outdated for the very recent Horizon 2, and possibly planning for a rebuild in the coming weeks.

Atik people, these are challenging times indeed and you're doing great products. :thumbsup:

knro, 2020-05-02:
I just emailed Atik again to check on this issue, hopefully we get a response soon.

TallFurryMan, 2020-05-08:
Pushed #123 for review and test in the context of this issue.

TallFurryMan, 2020-05-08:
Preliminary tests with the Horizon 2 are encouraging: the camera is detected and characteristics are retrieved properly. However sometimes it takes two attempts for the SDK to discover the device. I will push an update to PR #123 to retry the connection.

As a side note, Ekos will retry connecting a second time anyway, increasing the chance the device communication succeeds (but we know that Ekos does not reset the connection counter once connected).

TallFurryMan, 2020-05-10:
On my laptop running `Lubuntu 18.04 x64`, `libatik 2.2.7` (SDK v145) needs a second attempt to enumerate the Horizon 2 properly. This is taken into account in PR #123.
```
$ indiserver -p 7625 -vvv indi_atik_ccd
2020-05-10T19:26:15: startup: indiserver -p 7625 -vvv indi_atik_ccd 
2020-05-10T19:26:15: Driver indi_atik_ccd: pid=26093 rfd=3 wfd=6 efd=7
2020-05-10T19:26:15: listening to port 7625 on fd 4
2020-05-10T19:26:15: Driver indi_atik_ccd: sending msg copy 1 nq 1:
<getProperties version='1.7'/>

2020-05-10T19:26:16: Driver indi_atik_ccd: read message    'No Atik devices detected on attempt 1/5, retrying...'
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties Telescope Simulator EQUATORIAL_EOD_COORD 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties Telescope Simulator TELESCOPE_INFO 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties Telescope Simulator GEOGRAPHIC_COORD 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties Rotator Simulator ABS_ROTATOR_ANGLE 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties Focuser Simulator ABS_FOCUS_POSITION 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties CCD Simulator FILTER_SLOT 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on CCD Simulator.FILTER_SLOT
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties CCD Simulator FILTER_NAME 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on CCD Simulator.FILTER_NAME
2020-05-10T19:26:17: Driver indi_atik_ccd: read getProperties SQM SKY_QUALITY 
2020-05-10T19:26:17: Driver indi_atik_ccd: snooping on SQM.SKY_QUALITY
2020-05-10T19:26:17: Driver indi_atik_ccd: read defSwitchVector Atik Horizon 2 CONNECTION Idle rw
    CONNECT='Off'
 DISCONNECT='On'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defTextVector Atik Horizon 2 DRIVER_INFO Idle ro
 DRIVER_NAME='Atik'
 DRIVER_EXEC='indi_atik_ccd'
 DRIVER_VERSION='2.6'
 DRIVER_INTERFACE='6'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defNumberVector Atik Horizon 2 POLLING_PERIOD Idle rw
  PERIOD_MS='1000'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defSwitchVector Atik Horizon 2 DEBUG Idle rw
     ENABLE='Off'
    DISABLE='On'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defSwitchVector Atik Horizon 2 SIMULATION Idle rw
     ENABLE='Off'
    DISABLE='On'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defSwitchVector Atik Horizon 2 CONFIG_PROCESS Idle rw
 CONFIG_LOAD='Off'
 CONFIG_SAVE='Off'
 CONFIG_DEFAULT='Off'
 CONFIG_PURGE='Off'
2020-05-10T19:26:17: Driver indi_atik_ccd: read defTextVector Atik Horizon 2 ACTIVE_DEVICES Idle rw
 ACTIVE_TELESCOPE='Telescope Simulator'
 ACTIVE_ROTATOR='Rotator Simulator'
 ACTIVE_FOCUSER='Focuser Simulator'
 ACTIVE_FILTER='CCD Simulator'
 ACTIVE_SKYQUALITY='SQM'
```

TallFurryMan, 2020-05-10:
Tested OK with:
- Horizon 2 on `x64 Linux` and `x86 Linux`
- 314E on `x86 Linux`

No test feasible from my side on `armhf` until the PR is merged.

TallFurryMan, 2020-05-10:
As a side note, the second attempt to enumerate the Horizon 2 is also needed on `Lubuntu 18.04 x86`. However, the enumeration succeeds on first attempt for the 314E in the same environment.

TallFurryMan, 2020-05-17:
Before fixing this following #123, I need to observe two points.

In the observatory I need to control the power supply of the device 

When the power supply is cycled while the USB cord is still connected, the device is enumerated as:
```Bus 001 Device 002: ID 04b4:00f3 Cypress Semiconductor Corp.```
...which is not recognized by the Atik driver.

When the power supply is cycled then the USB cord is connected, the device is enumerated as:
```Bus 001 Device 005: ID 20e7:dfc3```
...which is properly recognized by the Atik driver..

This is strange and undocumented, but possibly a bootloader mode.
I'll ask support about it if this behavior is consistent.

TallFurryMan, 2020-05-20:
Atik Support replied:
```
The Horizon cameras use an FPGA, therefore the firmware is uploaded
each time the camera is power cycled. Consequently, you will hear the
windows connect sound followed, a few seconds later by the disconnect
sound, indicating that the firmware upload as completed.  

The Horizon I and II should both have the correct PID and VID when
they initially connect. However, the camera re-enumerating is a consistent
behaviour as we upload the firmware when our shared library is running.
```

I will test again the situation where the USB cord is connected before the power supply, and eventually open a new issue later.

Closing as resolved!

closed 2020-05-20 by TallFurryMan