indilib/indi-3rdparty #118: indi_mgenautoguider: Lacerts MGEN3 is not recognized

Weruminger, 2020-05-03:
**ISSUE:** MGEN3 is not recognized by the driver.

**SOLUTION:** MGEN3 shall be recognized by the driver.It will be okay, if not all features will run from the beginning,

minimum needed features. 
* MGEN3 shall be recognized and connected
* search stars 
* start guiding
* stop guiding

**Alternative workflow:**
Using [USB IP Bridge](http://usbip.sourceforge.net/) to remote controll with [Lacerta VirtualUI](https://support.mgen-autoguider.com/uploads/Virtual_UI_v10.exe) will work but is not processable in automation from the stratch.

**Support from my side**
Testing, log protocolls, _poor_ reverse engineering, using my MGEN3

TallFurryMan, 2020-05-05:
I attempted to do that in the past, but the result was relatively unstable. I wasn't able to commit on the result and abandoned development. The code is at https://github.com/TallFurryMan/indi, branch mgen_navigation. It will be difficult to merge and the commits should be picked as patches, but the workflow does just what you need. It is, as I said, very (very) easy to disturb and confuse. Ah, and this is for mgen2.

knro, 2021-09-28:
Ok marking this as "wontfix" since no update for a year now.

closed 2021-09-28 by knro