indilib/indi-3rdparty #124: Starbook pulse guiding

knro, 2020-05-08:
INDI Starbook driver does not currently support guiding as reported in the INDI forum thread [PHD2 can't manage to connect the mount](https://indilib.org/forum/general/6915-phd2-can-t-manage-to-connect-the-mount.html)

Is it possible to add this feature to the driver @not7cd ?

not7cd, 2020-05-08:
I could try reproducing this error. But I won't be able to contribute anything meaningful until July.

If I understand:
> Pulse guiding is an idea originally suggested by Roland Christen (founder of Astro-Physics) and implemented in this application. It works by sending precisely timed serial commands that make the mount move faster in RA and/or Dec for a short period (hence "pulses").

Then on the Starbook, it could be hard to achieve, because the driver uses quite limited HTTP API.

I was thinking about circumventing Starbook altogether and hooking directly into a serial port. But this will demand a lot of rev-eng.

knro, 2020-05-08:
one way would be to set it to GUIDE speed and then use regular NSEW commands with specific duration, would that work?

not7cd, 2020-05-08:
I guess, but I think it will be quite wonky in operation. But let the
evidence here pile up. And I can help with testing later since my access
to my observatory is limited due to the circumstances.

knro, 2021-09-28:
Following up a year after this! I guess this is still not supported?