indilib/indi-3rdparty #125: INDI driver for Brightstar Mammut Lyuba (or Orion Starshoot G3)

Manu4810, 2020-05-12:
I own an old Brightstar Mammut Lyuba camera which works fine on Windows with both native and ASCOM drivers (that are no more maintained since the company closed business). I would like to move my entire set-up to a full linux environement, with RaspberryPI / Astroberry, but unfortunately the camera is not recognized by the existing set of INDI drivers embedded in Astroberry (I tried MEADE DSI, Orion SSAG, Starshoot, all to no avail) and therefore cannot be used as main or guide camera. This camera ressembles the ORION Starshoot G3 very much. It seems to be based on some AVR32 AMTEL chipset from microchips (so it says from dmesg after plugging it into the USB port). Any advice on the relevant INDI driver to be used or any trick involving ASCOM Alpaca would be very much welcome. Thanks

knro, 2020-08-30:
So you confirm indi_starshootg_ccd driver does not work with it?

Manu4810, 2020-08-30:
I'm afraid I do.

knro, 2020-08-30:
Then there is no support for the camera, and given that there is no SDK and company closed, chances are quite slim unless you are willing to develop a driver for it.

bgilsrud, 2021-07-03:
@Manu4810 can you provide the USB VID and PID for this device? A few of us are working on a driver for the Orion Starshoot G3 if you care to join us: https://indilib.org/forum/general/878-support-for-orion-starshoot-g3-ccd.html

PR #430 (2021-08-12): Add driver for Orion StarShoot G3 cameras
This commit adds the indi_orion_ssg3_ccd driver which supports the Orion
StarShoot G3 cameras (color and mono).

fixes #125
closed 2021-08-12 by knro