indilib/indi-3rdparty #127: indi_astrolink4 : Focuser reverse motion has no effect

blemasle, 2020-05-13:
**Describe the bug**
Astrolink4's "Reverse motion" setting has no effect on focusing direction.

**To Reproduce**
1. Run driver
2. Focus In
2. Enable "Reverse Motion" under "Focuser" tab
3. Focus In

**Expected behavior**
The focuser motion should have been reversed.

**Desktop:**
 - OS: astoberry 2.0.1(raspbian 10 buster)
 - KStars 3.4.2, indilib 1.8.5

**Log Files**
Enabling reverse motion

> [2020-05-13T21:33:17.005 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] RES u:100000:250:0:100:500:1:0:0:446:3:1:1:0:0:1:0:0:2153:433: "
> [2020-05-13T21:33:17.005 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] CMD U:100000:250:0:100:500:1:0:0:446:3:1:1:0:0:1:0:0:2153:433:: "

Disabling reverse motion

> [2020-05-13T21:33:17.035 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] RES u:100000:250:0:100:500:0:0:0:446:3:1:1:0:0:1:0:0:2153:433: "
> [2020-05-13T21:33:17.035 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] CMD U:100000:250:0:100:500:0:0:0:446:3:1:1:0:0:1:0:0:2153:433:: "

The "reverse motion" setting, which according to [the code](https://github.com/indilib/indi-3rdparty/blob/master/indi-astrolink4/indi_astrolink4.cpp#L723) and [this define](https://github.com/indilib/indi-3rdparty/blob/master/indi-astrolink4/indi_astrolink4.h#L67) is the 6th item (the one after `500` should be `1` when enabled on the device.

The device itself and the reverse motion setting is working fine. If I set the "Reverse motion" from the Windows Astrolink panel, it's working as expected. Later when I plug the Astrolink into my INDI setup again, I can see that the setting is read as `1` before being updated to `0` again on load (`u` = read, `U` = write) :

> [2020-05-13T21:37:43.014 CEST DEBG ][   org.kde.kstars.ekos.capture] - Registering new Module ( "Capture" )
> [2020-05-13T21:37:43.033 CEST DEBG ][   org.kde.kstars.ekos.capture] - Registering new Module ( "Focus" )
> [2020-05-13T21:37:43.041 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] RES U: "
> [2020-05-13T21:37:43.042 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] CMD u "
> [2020-05-13T21:37:43.042 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] RES u:100000:250:0:100:500:1:0:0:446:3:1:1:0:0:1:0:0:2153:433: "
> [2020-05-13T21:37:43.042 CEST DEBG ][           org.kde.kstars.indi] - AstroLink 4 : "[DEBUG] CMD U:100000:250:0:100:500:0:0:0:446:3:1:1:0:0:1:0:0:2153:433:: "

So my guess is [bool IndiAstrolink4::ReverseFocuser(bool enabled)](https://github.com/indilib/indi-3rdparty/blob/master/indi-astrolink4/indi_astrolink4.cpp#L721) is always called with `enabled = false` no matter what the UI says. But I cannot find why looking at the code itself.

blemasle, 2020-05-13:
I can confirm that ReverseFocuser is always called with `enabled` set to `false`.
I'm able to fix the driver's behaviour by replacing the `ReverseFocuser` implementation with 
```cpp
bool IndiAstrolink4::ReverseFocuser(bool enabled)
{
    bool reversed = FocusReverseS[INDI_ENABLED].s == ISS_ON;
    return updateSettings("u", "U", U_REVERSED, (reversed) ? "1" : "0");
}
```

But I think the problem is wider than the astrolink4 driver and might affect other focusers as well. It is probably linked to the commit made by @knro [on April 20](https://github.com/indilib/indi/commit/df9760729845782dbe6a4daac435749ce41c7481#diff-21d78101ce94e484bc79271ccee70f07L469)

Shouldn't [this line](https://github.com/indilib/indi/blob/bef80d6b123a2d3b98b92cddd75c5730e47b8977/libs/indibase/indifocuserinterface.cpp#L478) be something similar to the Backlash activation instead ? Probably something in the essence of
```cpp
bool enable = !strcmp(FocusReverseS[DefaultDevice::INDI_ENABLED].name, IUFindOnSwitchName(states, names, n));

```

knro, 2020-05-14:
I simplified these lines further. Please update indi from git and then build the indi-astrolink4 driver and test.

blemasle, 2020-05-14:
It is indeed working as expected with indilib and indi-astrolink4 built against the latest sources. 👍

closed 2020-05-14 by blemasle