indilib/indi-3rdparty #140: Few nightscape issues

knro, 2020-05-26:
Jeff Donaldson raised a few issues regarding the camera on (Sourceforge)[https://sourceforge.net/p/indi/bugs/83/] and on (INDI Forum)[https://indilib.org/forum/ekos/7048-issues-with-flat-frames-and-the-celestron-nightscape-8300.html)

@dirkenstein Can you please take a look at these issues?

xxAstroJeffxx, 2020-05-26:
Thank you for the continued support @knro I really appreciate it.  This camera seems to work great in Windows in Sequence Generator Pro but in INDI its a nightmare.  60% of any frames I shoot are unusable, some 1/8th blacked out, some are half blacked out or green.  I am tried every USB port in the PI4, even a powered usb 2 hub.

You also compiled drivers so I can try the other NIghtscape 8300 driver.  sudo apt-get works and downloads and installs, confirmed by running the command again but its not an option to click on in the INDI control panel as per screenshots supplied by @dirkenstein on the Driver's page.

I am at a loss now and begging for help before I go completely bald! :)

Jeff

dirkenstein, 2020-05-26:
Hi,

Apologies for the slow reply- I haven't been working with this camera for
some time and mine is in storage atm so i can't necessarily reproduce the
issues you have quickly.

You've definitely added  the first line from this 99-nightscape.rules file
in /etc/udev/rules.d/?

ACTION=="add", ATTR{idVendor}=="19b4", RUN+="/bin/sh -c '/bin/echo 256
>/sys/module/usbcore/parameters/usbfs_memory_mb'"

# All Celestron Nightscape Cameras

SUBSYSTEMS=="usb", ATTR{idVendor}=="19b4", MODE="0666"

Without that, i've had all sorts of problems on raspberry pi systems.
Wouldn't have expected it on the pi 4, though.

Also, try a different cable, even if it works OK in Windows. I've noticed
these cameras are really sensitive to cable quality for some reason.

Are you using D2xx or libftdi driver backend?

The frame issues I will look at - the BGGR/RGGB issues are almost certainly
due to the fact that the frame downloaded from the camera is larger then
the visible frame size. It includes all sorts of blank and test pixels and
there may be a mistake in the cropping for that.

Regards,

Dirk

xxAstroJeffxx, 2020-05-26:
@dirkenstein thanks for the reply. Appreciate it. I am getting a new camera next year but this is what I have now. It’s been a solid performer. 

The coding part I have no idea what that’s for. I am not a coder 🙂

I have downloaded the drivers from StellarMates beta channel. Installed but it’s not selectable in the indi control panel. 

I will definitely try a new cable this evening and report back. 

Jeff Donaldson

dirkenstein, 2020-05-26:
Hi Jeff,

The coding part is a set of lines that need to be added (if not already
present) to a text configuration file on the raspberry pi. They change the
buffer memory for the USB interface.

This file is called 99-nightscape.rules and lives in the /etc/udev/rules.d/
directory. If it is not present it must be created.

If this is too difficult, you you can try running this command on its own
at the command line of the pi (you  need to log into it using ssh) :

sudo sh -c "/bin/echo 256 >/sys/module/usbcore/parameters/usbfs_memory_mb"

You need to do this _after_ you plug the camera in.

If you need help setting this up i can do some sort of screenshare or
similar via zoom or bluejeans or something to talk you through it.

There is a whole bunch of diagnostic stuff you can do to see if the camera
is recognized  but if you feel uncomfortable on the Stellarmate/raspberry
pi command line this is going to be pretty difficult to diagnose for me
without remote access to your hardware.

I've never tried the StellarMate OS or standalone product myself.

What were you running on when you got all the dropped frames- that was also
the StellarMate? I've had consistent problems on mac OS with this camera
and some better luck with Linux (again it needs those USB buffer tweaks to
work stably).

Regards,

Dirk

knro, 2020-05-26:
That file is already installed, so it's not the issue. What about libftdi1 vs libft2xx? The camera color array was verified? I noticed that was a "1" offset for Y?

dirkenstein, 2020-05-26:
Hi Jasem,

I'm sure I tested the colour array- and it should be set to BGGR in the
download. I think there's a bug in the driver where the reported bayer
pattern is RGGB but the fits file generator sets it correctly to BGGR.
Where did you see the 1 offset?

With the 99-celestron.rules file, libftdi1 should be as stable as libftd2xx
but it might stll be worth trying  libftd2xx because that seemed to
always work for me whereas libftdi1 was definitely 'tweaky'.

Regards,

Dirk

xxAstroJeffxx, 2020-05-26:
Thanks again. 

I saw rggb in all the spots it was mentioned. Even in the fits viewer. I had to tell the fits viewer to change the Bayer pattern each download. 

I saw it initially in the camera settings in the control panel. 

I did try to install the libftd2xx which did work but not seeing it on the control panel as a selectable option. Even after a reboot. 

I will head out later and try a different USB cable. I am using the Celestron Ac power adapter for the camera as well. 

Jeff

xxAstroJeffxx, 2020-05-27:
Ok. New cable. Same issue. Out of 20 subs 14 were usable. 

Resolution of the camera is 3326x2504 but the driver reports 3326x2506. I edit and save And keeps reverting and never reflects the saved res in camera module and sometimes even if forcibly changed doesn’t reflect the image and sometimes reports correctly And other times incorrectly. 

Here are a few images. 

[image]
[image]
[image]

knro, 2020-05-27:
@dirkenstein It's set [here](https://github.com/indilib/indi-3rdparty/blob/996dd94b0bbd03373b8b1d5002403b87decddf17/indi-nightscape/nightscape.cpp#L321) to RGGB with X:0 Y:1 offset. Is there a reason to do it this way?

What about the resolution issue report by @xxAstroJeffxx ?

dirkenstein, 2020-05-27:
Hi Jasem,

I think it was me fiddling until it looked like it worked. I suggest
changing it to BGGR 0 0 and seeing if we can get the resultant image to
align properly.

2506 vs 2504 is what the camera seemed to return over the interface- which
of the two lines should be dropped is unclear to me- we'd
need @xxAstroJeffxx to confirm if he's getting blank lines at the top or
bottom of the image.

I can check the returned data handling to see where it's going wrong in
this context but it's going to be tricky to debug  without the camera handy.

I need to do a little investigation to find out why setting the
resolution is not taking effect.

Regards,

Dirk

knro, 2020-05-27:
Alright, it's set now to BGGR with Offsets 0,0. Let me know if you need to me do change anything else.

xxAstroJeffxx, 2020-05-27:
Hey Guys.  I see there was a driver update and downloaded and rebooted my Pi4.  I am still seeing RGGB and 3326x3506 and under main control, libftdi is my only option.

knro, 2020-05-27:
@xxAstroJeffxx Try to update again now.

xxAstroJeffxx, 2020-05-27:
@knro Thanks, I checked for updates.  None available.  Do I need to be on the beta channel?

knro, 2020-05-27:
Yes, it's in beta.

xxAstroJeffxx, 2020-05-27:
@knro Thanks, I am now in Beta and ran package updater and no updates

knro, 2020-05-27:
@xxAstroJeffxx Are you sure you're in Beta? try again, the package should have been already updated and uploaded.

xxAstroJeffxx, 2020-05-27:
@knro ok, I see the drivers now and installed.  Prior to the install, Kstars won't load for some reason.  Its running 58c, pretty hot maybe I need to turn it off for a bit.  Its 33c outside right now.

xxAstroJeffxx, 2020-05-27:
@knro So I am not longer able to get Kstars to load, just an hourglass then nothing.  No error.  rebooted many times.

knro, 2020-05-27:
sudo apt-get -y install kstars-bleeding

xxAstroJeffxx, 2020-05-27:
@knro Thanks, ran that, said it was the latest version..  Still not running for some odd reason.  PHD loads on startup but Kstars just gives me an hourglass.

xxAstroJeffxx, 2020-05-29:
So far so good, new drivers are installed and I shot 20 test frames without Corruption.  I will test next clear night and report back.

xxAstroJeffxx, 2020-06-05:
Although it is much better, other issues arise such as Image Capture timed out.  Will not download any 900s Light frame.  Times out and starts over.  The resolution is incorrect for this camera.  I can force it in INDI but in the Camera Module, I have to manually adjust the resolution.  Align Module fails every time because it uses the wrong resolution and cannot be changed.

Not sure what else to do.  I may have to go back to my original windows setup until I can get a new camera in a few years.

knro, 2020-06-06:
@dirkenstein Any ideas on the resolution issue? or 900s second limit?

xxAstroJeffxx, 2020-06-06:
I am not sure if its just 900s but it does take longer for the failer due to the wait to finish the sub.  It's very frustrating to sit for 15mins to have the sub finish just fine but fail just before the download starts.  Celestron doesn't even support this camera on Windows anymore so I still feel RPI is my best bet to get everything working.  I am able to shoot 900s Darks but the download times out over all sub lengths not just 900s but more reliable around 600s (These are Narrowband)

I realize getting a more modern camera would fix everything but the camera I am looking for is expensive, especially to get one at the APS-C format.  I had plans to get a focus cube and field flattener/Reducer first and in a year or so, get a new camera.  Its been working great.  I have no issues really with the quality of pictures it puts out.

I appreciate everything you guys are doing!

xxAstroJeffxx, 2020-06-06:
[image]
These checkerboards show up in flats and lights now.  They are also in debayered images as well.

dirkenstein, 2020-06-06:
Just pushed some changes to the repo which should help with the number of
lines in the frame. They might alos help with the timeout- please test and
report any error messages you get on timeout.

I can't tell from the image whether it's corrupt after the first few lines
or the checkerboard has some other origin.

Regards,

Dirk

xxAstroJeffxx, 2020-06-11:
@dirkenstein I have tried to update from the indi site and I am getting this error:

aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Raspbian/buster

knro, 2020-06-11:
@xxAstroJeffxx The ones in INDI website do not work on buster or StellarMate, please update the beta channel in Stellarmate and you should get updated packages.

xxAstroJeffxx, 2020-06-11:
@knro Thanks, I am in Beta channel but only a couple of libraries were there.  Updated, thinking maybe the drivers are bundled with but nothing has changed.  I have been checking every hour for the last week :)

xxAstroJeffxx, 2020-06-19:
Alright, we have the new Drivers now.  No longer having any issues with the download but the resolution is still a bit weird.  In the Camera Module, it displays correctly but in the INDI control panel, it's still listed incorrectly at 3326x2506.  This messes with plate-solving for centering and leaves a 500-pixel black bar at the top of each image.

[image]

dirkenstein, 2020-06-19:
Hi Jef,

I'll check to see where the incorrect figures are coming from and let you
know if you need another updated version.

Regards,

Dirk

closed 2021-09-28 by knro