indilib/indi-3rdparty #142: Indi-pentax package recipe wrong?

karlrees, 2020-06-03:
There's still something wrong with the build for the indi-pentax package.  It's packaging the astrolink4 driver instead of the pentax driver.   

I'm still learning how all this packaging stuff works, but I think the source of the problem may be here: https://code.launchpad.net/~mutlaqja/+recipe/indi-pentax-nightly.   The recipe is pulling in the source code from the indi-astrolink4 directory instead of indi-pentax.  Didn't see a way for me to change the recipe or propose changes.

karlrees, 2020-06-06:
@knro  - Any chance you have some time to look at the recipe for the package?  Or is there a way for me to propose changes to it?

knro, 2020-06-06:
Thanks for spotting that! It was a stupid copy/paste mistake. Fixed!

closed 2020-06-06 by knro