indilib/indi-3rdparty #150: indi-webcam not being built although FFMPEG is found

schigum, 2020-06-12:
**Describe the bug**
When compiling indi-3rdparty the webcam component does not get configured by cmake and thus does not get compiled. The compilation is done as advised in the README, starting with
`cmake  /path/to/source`
The output then implies indi-webcam would be built:
`-- Since FFMPEG was found, INDI Webcam Driver can be built`
However, the indi-webcam directory in the build folder is not created and cmake -L confirms that something is not as expected:
`WITH_WEBCAM:BOOL=OFF`

I'm surprised though, that I seem to be the first person having this problem, so I expect I'm doing something wrong...

I'm not all that familiar with the internals of cmake, so this might not be a "good" solution, but a small change to the main CMakeLists.txt produces the expected result:
changing `option(WITH_WEBCAM "Install INDI Webcam Driver based on FFMPEG" On)`
to `option(WITH_WEBCAM CACHE BOOL "Install INDI Webcam Driver based on FFMPEG" On)`
I have attached a patch applying this change, however I don't know if this has any unwanted side effects.
[patch.txt](https://github.com/indilib/indi-3rdparty/files/4772142/patch.txt)

**To Reproduce**
1. Have FFMPEG properly set up, such that it is found by cmake
2. From an empty directory, call cmake /path/to/src/
3. Check that FFMPEG was found and whether the indi-webcam folder is created

**Expected behavior**
If FFMPEG is available an indi-webcam folder should be created in the build folder.

**Desktop**
 - OS: Manjaro and Raspbian
 - Git master
 - cmake 3.17.3 (Manjaro, can't check for Raspian right now)

knro, 2020-06-13:
@rlancaste can you please check? I thought this was fixed recently?

knro, 2020-07-08:
Can you check if this issue resolved? @rlancaste @schigum

closed 2021-09-28 by knro