indilib/indi-3rdparty #151: indi-gphoto indi_sony_ccd Sony a5000 subsequent exposures not captured

m-kostrzewa, 2020-06-13:
**Describe the bug**

I have Sony Alpha A5000 (ILCE-5000).

Only about every 3rd exposure done by Ekos using indi_sony_ccd driver actually moves the shutter and captures new data. Other exposures don't cause the shutter to open or close, and stale data is downloaded as preview (from the successful exposure).

I have investigated this issue and provide a hack/patch that works for me.

Additional keywords: freeze, frozen, stuck, 

**To Reproduce**

Reproduce it using Kstars+Ekos.

In profile editor, add Sony DSLR device. I have Sony Alpha a5000 (ILCE-5000) connected via USB in PC direct mode.
Run the profile.

1. Go to CCD tab.
2. Grab preview exposure. Preview should appear.
3. Move the camera so that it points to a different object (to compare previews).
4. Grab preview exposure. You will notice there's no shutter sound, and the preview is the same as in step 2.
5. Repeat steps 3-4 until a "fresh, real" preview appears (around 3 times).

Note, that no error is reported here.

This may or may not be relevant, but I also disabled LENR via telnet like so: https://github.com/ma1co/OpenMemories-Tweak/issues/205 This requires installing OpenMemories Tweak app on the camera itself. I initially suspected it to be the problem, but even though this hack works (LENR is disabled), it didn't solve this issue. I'm not sure if it is required for my patch to work.

During the investigationI ran some gphoto2 commands to try to replicate the issue manually: 

```bash
astroberry@astroberry:~/codez/indi-3rdparty/indi-gphoto/build $ gphoto2 -v --set-config bulb=1
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

astroberry@astroberry:~/codez/indi-3rdparty/indi-gphoto/build $ gphoto2 -v --set-config bulb=0 --wait-event-and-download=5s --force-overwrite
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.24.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.           
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Event c201, Param1 ffffc001
Saving file as capt0000.arw                                                    
Saving file as capt0001.arw                                                    
```

You may notice, that even though I grab only one image, two files are downloaded (capt0000.arw and capt0001.arw). Sometimes it is1 file, sometimes 3, depending on the sequence of actions before. I can also run another pair of commands like above (bulb=1/bulb=0) immediately after without it getting stuck. This made me suspect that the issue is not withing libgphoto2, but with the indi driver. I thought that maybe the indi driver doesn't handle the situation where there are multiple files to be downloaded.

I have no idea why multiple files are generated to begin with. If someone could enlighten me, I'd appreciate it.

**Expected behavior**
[... 175 lines cut ...]

[log_patched.txt](https://github.com/indilib/indi-3rdparty/files/4775422/log_patched.txt)

After applying the aforementioned patch: all the files to be downloaded (GP_EVENT_FILE_ADDED events) are 'drained' until GP_EVENT_CAPTURE_COMPLETE. The last file's data is the one passed to the caller. Driver works as expected.

I was inspired by this comment: https://github.com/gphoto/gphoto2/issues/48#issuecomment-294225211

Excerpt:

```
DEBUG	9.271595 sec	: Exposure complete.
DEBUG	10.450409 sec	: Event timed out #1, retrying...
DEBUG	12.417993 sec	: File added event completed.
DEBUG	12.418198 sec	: Deleting aborted image... Name: (capt0000.arw) Folder: (/)
DEBUG	12.530922 sec	: Could not determine image size (Unspecified error)
DEBUG	14.373861 sec	: File added event completed.
DEBUG	14.374023 sec	: Deleting aborted image... Name: (capt0001.arw) Folder: (/)
DEBUG	14.483128 sec	: Could not determine image size (Unspecified error)
DEBUG	16.224815 sec	: File added event completed.
DEBUG	16.224969 sec	: Deleting aborted image... Name: (capt0002.arw) Folder: (/)
DEBUG	16.334314 sec	: Could not determine image size (Unspecified error)
DEBUG	16.344237 sec	: Capture event completed.
DEBUG	72.498099 sec	: Requested CCD Frame is (0,0) (5456 x 3632)
INFO	72.498755 sec	: Starting 1 seconds exposure.
```

msmeissn, 2020-06-14:
I think the event handling in indi should cope with 1 or more images being added, which can happen if you do triggering outside of the usb protocol, (or even if someone presses the shutter button).

knro, 2020-06-14:
@m-kostrzewa Thank you for the detailed report!

@msmeissn I'm trying to do that now, but it appears that some cameras (e.g. my Canon 600D) the event GP_EVENT_CAPTURE_COMPLETE is never received while it is triggered for Nikon 3500D. Can you please check [gphoto_read_exposure_fd](https://github.com/indilib/indi-3rdparty/blob/821b83fc3d5421d964f6175dfa38db1b8e0aac3b/indi-gphoto/gphoto_driver.cpp#L1305) and see what we're doing wrong? Right now the function returns after GP_EVENT_FILE_ADDED which works for the majority of cameras.

knro, 2020-06-14:
@m-kostrzewa Can you please apply this patch and see if it works in your case?

[gphoto.zip](https://github.com/indilib/indi-3rdparty/files/4776263/gphoto.zip)

msmeissn, 2020-06-14:
CAPTURE_COMPLETE is not reliable sadly. I would poll events with a short timeout and see if I get a GP_EVENT_TIMEOUT timeout back if I not expecting more pictures.

knro, 2020-06-14:
@msmeissn This is exactly what I did in the patch above! Will wait for @m-kostrzewa to test it out. Thank you for the feedback.

m-kostrzewa, 2020-06-14:
@knro thanks for swift reply. I tested your patch on sequences of 1sec and 15sec exposures. Works great.

knro, 2020-06-14:
@m-kostrzewa does it work for 30+ sec exposures as well?

msmeissn, 2020-06-14:
> @msmeissn This is exactly what I did in the patch above! Will wait for @m-kostrzewa to test it out. Thank you for the feedback.

Ah yes, looks good. :)

m-kostrzewa, 2020-06-14:
@knro tested on sequence of 60 sec exposures, it works

m-kostrzewa, 2020-06-15:
@k-ross tested it tonight on sequences of 180, 300, 600sec exposures. I've encountered some hiccups. Good news is that the happy path works great, and I think this bug is resolved.

The hiccups I encountered were mostly "sad path" edge cases. Some were encountered because I prefer to focus the camera using mobile phone WiFi app. When I'm done I turn the app off and reconnect INDI driver. There were also instances where aborts/pauses made the camera "stuck".  I haven't debugged it too much nor marked the logs in any way though...  When I encounter these issues again I will report them properly. Here's my memory dump:

* Disconnecting/connecting the camera via Ekos sometimes works, sometimes doesn't with error message to check the cable. May be user error on my part.
* Sometimes the first preview exposure after reconnecting to INDI driver from Ekos is stuck in "Downloading", even though it appears in the preview. **Maybe this suggests the event handling routine can be made more resilient?** (I will need to capture better logs for this as I don't know how to reproduce).
* one time, I had error where driver couldn't connect to the camera at all, I had to restart everything (kstars+ekos, indi drivers, camera power cycle). 
* This was encountered also before patching, but almost always when connecting INDI sony driver via Ekos, an exposure is immediately started. There's an easy workaround - just abort it once after connection is estabilished via Ekos GUI. I've no idea why this happens though.

But as I said, I think this bug is mostly resolved. Thanks.

kleb0nas12, 2020-12-16:
Hi guys , If someone would be able to explain how to 'patch' this problem that would be great! Because I am facing exact the same problem (with sony alpha a5000) and the only thread I found on internet is here. The problem is the same: ekos only donwloads every second photo and sometime , on start hangs with the first one.

glundby, 2020-12-16:
Long time since I did the hack but here is the links I used then:
https://github.com/ma1co/Sony-PMCA-RE/releases/tag/v0.17
https://github.com/ma1co/OpenMemories-Tweak/issues/169
Happy hacking :-)

m-kostrzewa, 2020-12-17:
@kleb0nas12 

I built it manually months ago and can't remember the details, but the overall structure of what needs to be done is as follows

1. Download .zip file uploaded by @knro 
2. Need to clone (using git) all some indi repositories in order to build indi drivers (can't remember which, sorry)
3. Use `git apply` to apply code patch from the .zip file
4. Download indi build dependencies (need to go through relevant README files in some repo)
5. Build relevant indi driver (can't remember which, but filename in .zip should help) using `make asdfasfasdf` command. This is also described in some README file in some repo.
6. Find the old driver binary, backup it, replace with the one you've built.

It's kinda cumbersome. If you have software development/IT experience it should help. Even then, building such big project for the first time is often frustrating :) and takes some time. If you hit any road blockers, don't be afraid to ask

Ideally, @knro merges his patch and it is released officially in newer version soon. But I imagine it's a breaking change for some other camera models. @knro what is stopping you from merging your patch?

m-kostrzewa, 2020-12-17:
Actually, as https://github.com/indilib/indi-3rdparty/issues/230 noticed, this patch is already merged to main branch.

So just needs to download the newest version of indi driver, @kleb0nas12 

@knro may I close this issue?

kleb0nas12, 2020-12-17:
I am using up-to-date indi drivers (sudo apt update doens't update indi drivers?)

knro, 2020-12-18:
Right INDI packages already have this in. Check that you have INDI v1.8.7 installed.

closed 2020-12-18 by knro

peppuccio, 2020-12-21:
Hello,
I have the same problem with my Sony A7S. I'm new to open solutions and run an astroberry version on PI 4 under Raspberry OS. 
How can I be sure that my Indi version is the good one?

Thanks
J. Ferraro
Paris

m-kostrzewa, 2020-12-22:
Show `apt list --installed`, look for indi

If version is too low, try

```
sudo apt-get update && apt-get upgrade
```

If version is still too low, report back. I haven't tried this yet :)

peppuccio, 2020-12-22:
Thanks for you response... the first one since I try to solve this problem !
here is a capture for apt list --installed 
[image]

And the same command after apt-get update && apt-get upgrade
[image]
It's sounds to me that I don't have 1.8.7 version in both case

G. Ferraro

m-kostrzewa, 2020-12-22:
Hmm.. maybe newest version is not in apt repository yet...

In this case, you must build the software yourself.

peppuccio, 2020-12-22:
Thanks a lot.
I put a request in the Astroberry section of the Indi Forum. I would prefer an official package... 
We have a forecast of 10 days raining here in Paris... and If needed I can use an ascom driver under widows!

G. Ferraro

kleb0nas12, 2021-01-01:
Yes indeed. I reinstalled INDI on my raspberry pi , and it still 1.8.6

peppuccio, 2021-01-02:
I open an issue in the astroberry section of the INDI forum, but I receive no reply.
Perhaps it is due to the new year vacations?

G. Ferraro

kleb0nas12, 2021-01-14:
I don't know, but there is still the same 1.8.6 version.