indilib/indi-3rdparty #153: QHYCCD QHY183M LiveView not working

MOS6522, 2020-06-19:
When using the Streaming in INDI QHY Driver with a QHY183M, the stream fails and often locks up KStars if left alone too long.

1. Define QHYCCD as main CCD Driver
2. Start KStars
3. Open INDI Control Panel and select the QHY183M
4. Change to Streaming Tab
5. Turn Streaming On and wait
6. Image will be black and log will have streaming errors

It would be nice if Streaming worked on QHY183M cameras

I have found the issue. It appears to be around the discrepancy between the call to PrimaryCCD.getFrameBufferSize() (which returns 40MB) and the call to GetQHYCCDLiveFrame, which returns params that would indicate a 20MB Buffer is needed.

Just as a test, I compiled the code with a lot more debugging, showing sizes of expected buffers and actual buffers. They were off, the expected was 40MB and the reality was 20MB. I simply divided the expected by 2 in the call to Streamer->newFrame and it started to work. This is a hach, for sure, but it does show there is something off.

JBielanski, 2020-09-02:
I have similar problem with QHY600M camera in video mode. I modified a bit **qhy_video_test** program code, now the bug is detected in test program. That patch not solve the problem, but return more information. To apply patch run code: 
`patch < qhy_video_test.patch indi-3rdparty/indi-qhy/qhy_video_test.cpp`
**qhy_video_test** result:
`qhy_video_test` 
`QHY Video Test using VideoFrameMode, Version: 1.00`
`SDK resources initialized.`
`Number of QHYCCD cameras found: 1`
`Application connected to the following camera from the list: Index: 1,  cameraID = QHY600M-a7ca3c438f7ea8e27`
`Open QHYCCD success.`
`InitQHYCCD success.`
`GetQHYCCDOverScanArea:`
`Overscan Area startX x startY : 0 x 0`
`Overscan Area sizeX  x sizeY  : 10 x 6388`
`GetQHYCCDEffectiveArea:`
`Effective Area startX x startY: 0 x 0`
`Effective Area sizeX  x sizeY : 10 x 6388`
`GetQHYCCDChipInfo:`
`Effective Area startX x startY: 0 x 0`
`Chip  size width x height     : 36.096 x 24.147 [mm]`
`Pixel size width x height     : 3.760 x 3.760 [um]`
`Image size width x height     : 9600 x 6422`
`This is a mono camera.`
`SetQHYCCDParam CONTROL_EXPOSURE set to: 1 us, success.`
`SetQHYCCDParam CONTROL_USBTRAFFIC set to: 20, success.`
`SetQHYCCDParam CONTROL_GAIN set to: 1, success`
`SetQHYCCDParam CONTROL_GAIN set to: 180, success.`
`SetQHYCCDResolution roiStartX x roiStartY: 0 x 0`
`SetQHYCCDResolution roiSizeX  x roiSizeY : 9600 x 6422`
`SetQHYCCDBinMode set to: binX: 1, binY: 1, success.`
`Allocated memory for frame: 126526800 [uchar].`
`Press any key to exit...`
`[videoThread] GetQHYCCDLiveFrame: [w=9600 / h=6422 / bpp=8 / channels=1] error code: 4294967295 !!!`
`Close QHYCCD success.`
`SDK resources released`
Patch file:
[qhy_video_test.patch.zip](https://github.com/indilib/indi-3rdparty/files/5162095/qhy_video_test.patch.zip)

knro, 2020-09-02:
So this looks like an SDK issue that is best communicated back to QHY.

JBielanski, 2020-09-02:
I modified code a bit more and I received following data for POLEMASTER:
`qhy_video_test `
`QHY Video Test using VideoFrameMode, Version: 1.00`
`SDK resources initialized.`
`Number of QHYCCD cameras found: 1 `
`Application connected to the following camera from the list: Index: 1,  cameraID = POLEMASTER-0047e910b78ae44eb`
`Open QHYCCD success.`
`InitQHYCCD success.`
`GetQHYCCDOverScanArea:`
`Overscan Area startX x startY : 0 x 0`
`Overscan Area sizeX  x sizeY  : 0 x 0`
`GetQHYCCDEffectiveArea:`
`Effective Area startX x startY: 0 x 0`
`Effective Area sizeX  x sizeY : 0 x 0`
`GetQHYCCDChipInfo:`
`Effective Area startX x startY: 0 x 0`
`Chip  size width x height     : 4.800 x 3.600 [mm]`
`Pixel size width x height     : 3.750 x 3.750 [um]`
`Image size width x height     : 1280 x 960`
`This is a mono camera.`
`SetQHYCCDParam CONTROL_EXPOSURE set to: 1 us, success.`
`SetQHYCCDParam CONTROL_USBTRAFFIC set to: 20, success.`
`SetQHYCCDParam CONTROL_SPEED set to: 2, success.`
`SetQHYCCDParam CONTROL_GAIN set to: 1, success`
`SetQHYCCDResolution roiStartX x roiStartY: 0 x 0`
`SetQHYCCDResolution roiSizeX  x roiSizeY : 1280 x 960`
`SetQHYCCDBinMode set to: binX: 1, binY: 1, success.`
`Allocated memory for frame: 2925600 [uchar].`
`Press any key to exit...`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] Frames: 10 Duration: 3.108 seconds FPS: 3.217`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] Frames: 12 Duration: 3.110 seconds FPS: 3.859`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] Frames: 12 Duration: 3.124 seconds FPS: 3.841`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`[videoThread] Frames: 12 Duration: 3.112 seconds FPS: 3.856`
`[videoThread] GetQHYCCDLiveFrame: [w=1280 / h=960 / bpp=8 / channels=1] error code: 4294967295 !!!`
`Close QHYCCD success.`
`SDK resources released.`
So error appear when program can't get a frame or frame is download from camera. 
I attached new patch version:
[qhy_video_test.patch.zip](https://github.com/indilib/indi-3rdparty/files/5162434/qhy_video_test.patch.zip)

commit 62f00c21 (2020-09-06): Correct size for video stream.

Fixes #153
closed 2020-09-06 by knro