indilib/indi-3rdparty #156: Starlight Xpress not working in ekos

piggled, 2020-06-23:
I am connecting my SX-56 ccd to ekos/kstars. I run an exposure and it times out. Camera is connected, cooler is on, shutter opens and closes, ekos counts down, clock stops at 0 and stalls.  No error codes pop up on indi and it is stalled for a couple minutes before it tries attempt number 2.  However, in the platesolving module the ccd will expose and the mount will slew. It is just not working in the capture module. I have connected dozens of other cables, even wiped my entire computer to start from new. In the platesolver it is only connected to the SX-56 as that is the only camera I have on the mount right now.  I am running the latest Kstars on a mac book pro with catalina. 

I did just realize that the camera is working in the solver because it si binned at 2x. the camera will not work when it is not binned. Please help! 

P.S. I am not a coder. I am just a normal dude looking to use his camera. thanks!

bluthen, 2020-06-24:
I recently(8 months back) had a pull request to increase the timeout for an older starlight camera because data transfer was slower in that model. I wonder with this camera, when no binning, if the driver is just not waiting long enough to get all the data. How long would you say it takes to get the image from the camera when you do 2x2 bining?

#11

piggled, 2020-06-24:
2x2 it pops up in around 6seconds. This is a brand new SX-56. The download speeds are supposed to be around 7s or so for a full image. It works in TSX pro. Just not in Kstars. In TSX download time is around 7-8s 

Ryan

bluthen, 2020-06-24:
If 2x2 takes 6s a 1x1 may end up taking 4*6s = 24s. The current timeout is set to 20s.

Not sure why it would take so much longer than TSX. But I suspect that is the issue. 

24s between each frame can be a lot of darktime you lose out on. When you tried in TSX pro vs INDI, is it the same USB chain of cables or hubs or whatever?

piggled, 2020-06-24:
Yes. All the cabling is the same. How do I change the timeout time? I don’t care about the lost time. Just want to use the full resolution and really do not like TSX as a program. 

Ryan

bluthen, 2020-06-24:
Have to recompile the indi_sx_ccd file after changing indi-sx/sxccdusb.cpp:

`
#define BULK_DATA_TIMEOUT    20000 //Older SXV-M25C takes 14s unbinned
`
To something like 40000

piggled, 2020-06-24:
Shoot. Ok. That is way above my pay grade. 

Ryan

bluthen, 2020-06-24:
I'm not an official developer, but maybe I can push attach a PR to make it 40000 and you can get it in next release? 

A real fix would be to figure out why it takes so long though.

piggled, 2020-06-24:
That would be awesome. Let me actually see how long the 2x takes. It is a medium format ccd. SX says download should only take 7s unbinned though. 

Ryan

closed 2021-09-28 by knro