indilib/indi-3rdparty #163: indi-rpicam crash

rwrw01, 2020-07-06:
on raspian buster with HQ cam
added indi-rpicam, build went well

on connecting to camera a warning fires:
2020-07-06T19:22:32: [WARNING] Connect: Unknown camera name: _testc_ 
2020-07-06T19:22:31: [INFO] MMAL device connected successfully!
 
no other options are available. When starting device it crashes

knro, 2020-07-07:
@lboclboc Can you please check?

lboclboc, 2020-07-08:
I recognize this, can you please do and sudo apt-get update and sudo apt-get upgrade. I think the problem is in the gpu driver really. Updated softwares will correctly report the camera as imx477. I'll also create a patch in the driver so it also accepts the faulty camera model testc

rwrw01, 2020-07-09:
I updated RPi on installation and did so again just now. Same error unfortunately.

lboclboc, 2020-07-09:
Ive created a pullrequests from https://github.com/lboclboc/indi-3rdparty/tree/indimaster that should fix this. Can you please try this. Seems later mmal libraries actually fixes the name, but this fix just simply accepts the bad name testc also.

rwrw01, 2020-07-11:
the camera is found.

now the next step how to apply gain/binning but that is probably my problem with the ekos interface

lboclboc, 2020-07-12:
Good to hear. Binning is not supported yet but gain is working.

kduchrow, 2020-09-10:
Hi, i am getting the same Error with an different camera name : 
2020-09-10T21:37:18: [WARNING] Connect: Unknown camera name: imx219
2020-09-10T21:37:18: [INFO] MMAL device connected successfully!

Thank you hor your help !

lboclboc, 2020-09-11:
Im sorry but this driver only supports the hi-quality camera yet. But I can have a look into supporting this camera. I need to find som info about its raw encoding though, if anyone have any references for that, please share with me.
Alternatively, I might be able do deduce the format I you can send me a "raspistill -r -o sample.jpg" capture.

kduchrow, 2020-09-11:
Unfortunately I do not know exactly what you need. But I have two links here that might help you: 
https://www.raspberrypi.org/forums/viewtopic.php?t=146310
http://bealecorner.org/best/RPi/

lboclboc, 2020-09-12:
Thanks, that is good input. I just realized I do have a IMX219 camera also, broken lens, but still I can examine the image format. To be continued.

kduchrow, 2020-09-13:
Awesome! 
Sounds very good. I think a lot of people already got a similar camera.
It would be very helpful to support such camera models. 
If you need more information, help, or testing, feel free to contact me.

lboclboc, 2020-09-13:
Actually I might have something working now. If you can help me test it I would appreciate. Use my fork on https://github.com/lboclboc/indi-3rdparty
Regards, Lars

kduchrow, 2020-09-13:
Thank you for dealing with this !
I tried the new driver and it istantly crash.
Here you can see my logfile:

```
2020-09-13T22:09:00: startup: /usr/bin/indiserver -v -p 7625 -m 300 -r 0 -f /tmp/indififo2d9658af 
2020-09-13T22:09:00: listening to port 7625 on fd 3
FIFO: start indi_rpicam -n "RPI Camera"
With name: RPI Camera
FIFO: Starting driver indi_rpicam
2020-09-13T22:09:00: Driver indi_rpicam: pid=6326 rfd=4 wfd=7 efd=8
2020-09-13T22:09:00: Client 5: new arrival from 127.0.0.1:47716 - welcome!
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_INFO
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2020-09-13T22:09:00: Driver indi_rpicam: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2020-09-13T22:09:00: Driver indi_rpicam: snooping on CCD Simulator.FILTER_SLOT
2020-09-13T22:09:00: Driver indi_rpicam: snooping on CCD Simulator.FILTER_NAME
2020-09-13T22:09:00: Driver indi_rpicam: snooping on SQM.SKY_QUALITY
2020-09-13T22:09:07: Driver indi_rpicam: mmal: mmal_vc_port_enable: failed to enable port vc.ril.image_encode:in:0(OPQV): ENOSPC
2020-09-13T22:09:07: Driver indi_rpicam: mmal: mmal_port_enable: failed to enable connected port (vc.ril.image_encode:in:0(OPQV))0x6a7300 (ENOSPC)
2020-09-13T22:09:07: Driver indi_rpicam: mmal: mmal_connection_enable: output port couldn't be enabled
2020-09-13T22:09:07: Driver indi_rpicam: terminate called after throwing an instance of 'MMALException'
2020-09-13T22:09:07: Driver indi_rpicam:   what():  Failed to enable connection
2020-09-13T22:09:07: Driver indi_rpicam: stderr EOF
<delProperty device="RPI Camera"/>
2020-09-13T22:09:07: Driver indi_rpicam: Terminated after #0 restarts.
Child process 6326 died
FIFO: stop indi_rpicam -n "RPI Camera"
With name: RPI Camera
dp->name: indi_rpicam - tDriver: indi_rpicam
FIFO: start indi_rpicam -n "RPI Camera"
With name: RPI Camera
FIFO: Starting driver indi_rpicam
2020-09-13T22:09:12: Driver indi_rpicam: pid=6404 rfd=4 wfd=8 efd=9
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_INFO
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2020-09-13T22:09:12: Driver indi_rpicam: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2020-09-13T22:09:12: Driver indi_rpicam: snooping on CCD Simulator.FILTER_SLOT
2020-09-13T22:09:12: Driver indi_rpicam: snooping on CCD Simulator.FILTER_NAME
2020-09-13T22:09:12: Driver indi_rpicam: snooping on SQM.SKY_QUALITY
2020-09-13T22:09:15: Driver indi_rpicam: mmal: mmal_vc_port_enable: failed to enable port vc.ril.image_encode:in:0(OPQV): ENOSPC
2020-09-13T22:09:15: Driver indi_rpicam: mmal: mmal_port_enable: failed to enable connected port (vc.ril.image_encode:in:0(OPQV))0x1d412a0 (ENOSPC)
2020-09-13T22:09:15: Driver indi_rpicam: mmal: mmal_connection_enable: output port couldn't be enabled
2020-09-13T22:09:15: Driver indi_rpicam: terminate called after throwing an instance of 'MMALException'
2020-09-13T22:09:15: Driver indi_rpicam:   what():  Failed to enable connection
2020-09-13T22:09:15: Driver indi_rpicam: stderr EOF
<delProperty device="RPI Camera"/>
2020-09-13T22:09:15: Driver indi_rpicam: Terminated after #0 restarts.
Child process 6404 died
FIFO: stop indi_rpicam -n "RPI Camera"
With name: RPI Camera
dp->name: indi_rpicam - tDriver: indi_rpicam
```

kduchrow, 2020-09-13:
Okay my Fault ! 
I had the Web UI in parallel working and probably blocking.
I stopt the Web UI for Raspi Cam and now the Camera is connecting instantly. I am going to test it further tomorrow .
Tank you !

lboclboc, 2020-09-14:
That is strange, can you upload the fits file you think? Also, do you have the automatic debayer option turned on i kstars?

kduchrow, 2020-09-14:
No i haven't. I turned it on, and i looks like it is working. 
I haven't enought time to get a proper fokus, but I have two testimages. 
I will test it further today eve.
[Test_Images.zip](https://github.com/indilib/indi-3rdparty/files/5216976/Test_Images.zip)

lboclboc, 2020-09-14:
Ok, in the second image I can clearly see the bayer raster indicating that the image is really being fetched from the camera ok but its very dark. Can you try to increase the exposure time and/or gain perhaps?
(sorry for asking but of course you don't have the lens cap on?)

kduchrow, 2020-09-14:
Hi,
yes I removed the lens cap. 
I had some trouble with the exposure setting. But I got a test picture that might help.
[Light_001.zip](https://github.com/indilib/indi-3rdparty/files/5218109/Light_001.zip)

Changing exposue time and gain looks also good. It is not possible to change the ISO is this right ?
Furthermore the image is only in grayscale or do i using the wrong viewer or do I need to configure it?

P.S. : It would be very nice if there is a liveview possible maybe in lower resolution?

lboclboc, 2020-09-14:
The reason you image is grayscale is probably because you have not turned on the "automatic debayer" in kstars, so you see the BGGR pixels with they values in grayscale only. 
I agree about liveview but I am not sure how that should work with INDI and kstars. The uploading of a whole raw picture for sure takes a long time.

lboclboc, 2020-10-18:
I think this ticket can be closed now.

closed 2021-09-28 by knro