indilib/indi-3rdparty #167: build community RPM packages for other distributions from git in an easy way

xsnrg, 2020-07-19:
**Problem**: As a user of a non-debian based system, I would like to easily manage and update installs of indi-3rd party drivers on my systems. My distribution does not provide an option for nightly builds, and this is a desirable outcome.

**Outcome**: I would like for the community to have a way to enjoy nightly builds and debugging on distributions other than debian based. My own person usage is for Fedora, but this should also encompass RHEL, CentOS, and SUSE if possible

**Alternatives:** I have spent a great deal of time looking at build engines, packaging methods, including AppImage, considering copying what Christian had done before me, and have settled on the Fedora Copr system

It would be desirable to simply be able to add the indi-3rdparty-bleeding repo and use system based tools such as yum, dnf, rpm, etc to do package management of installed versions.

knro, 2020-08-13:
What's the update on this? Any PR planned for this?

xsnrg, 2020-08-13:
The direction forward on this is up for discussion. We could include the spec files that I currently have in the xsnrg/*-spec repos in upstream (indilib/indi-3rdparty), and then reference them in the COPR builds. This would allow us to use webhooks directly to rebuild packages on commits to master. That would also remove the manual step of updating and pushing my -spec repositories to rebuild the RPM packages.

proposed process:
Change in github indilib/indi-3rdparty -> webhook to copr -> new package is built and made available at copr

Today the process is:

Change in indilib/indi-3rdparty -> manual update to xsnrg/indi-3rdparty-spec -> push to build copr packages

If we make the spec files part of indilib/indi-3rdparty, I would also want to do this for indilib/indi.

I hope I explained this clear enough. The question is if the maintainers want the spec files to be part of the main git repository.
I would then want to be allowed to be a maintainer in indilib/* to keep things up to date as needed.

Thoughts?

knro, 2020-08-13:
we already have debian files in the repo, so adding spec files shouldn't be an issue.

xsnrg, 2020-08-13:
Sounds good, I will work on a PR.

xsnrg, 2020-11-13:
closed by #226

closed 2020-11-13 by xsnrg