indilib/indi-3rdparty #177: indi_canon_ccd 3.0 driver crashes after capture before file transfer

BenjaminMerget, 2020-07-30:
**Describe the bug**
Using my Canon 2000D with indi_canon_ccd (or indi_gphoto_ccd) version 3.0 in kstars causes the driver to crash and restart after capture or preview during file transfer. Live view is not affected. On two different machines running indi_canon_ccd 2.9 and 2.8, respectively, there are no issues. (Edit: Same issue with different Canon DSLR (400D)).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect Canon 2000D two computer
2. Start Kstars with Canon DSLR driver (indi_canon_ccd 3.0 loaded)
3. Shoot preview image

**Expected behavior**
The image being transferred to the computer and opened in a viewer.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - kstars-bleeding and indi-full from stable release as well as nightly build

**Log Files**
org.kde.kstars.ekos.capture: Preparing capture job "/home/benjamin/Light" for execution.
org.kde.kstars.indi: Requested exposure 1 closes match is 1
org.kde.kstars.ekos.capture: "Capturing 1,000-second  image..."
org.kde.kstars.indi: Canon DSLR EOS 2000D :  "[INFO] Starting 1 seconds exposure. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Starting exposure (exptime: 1 secs, mirror lock: 0, force bulb: true, exposure index: 1869115495) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Mutex locked "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting radio/menu widget iso: 5 (1600) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting new configuration OK. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting radio/menu widget imageformat: 9 (RAW) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting new configuration OK. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting exposure widget bulb index: 0 "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting radio/menu widget shutterspeed: 0 (bulb) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting new configuration OK. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Using internal bulb widget:eosremoterelease "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting radio/menu widget eosremoterelease: 2 (Press Full) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting new configuration OK. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Exposure started. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Time left: 999 ms "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Time left: -1 ms "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Closing internal shutter. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Using widget:eosremoterelease "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting radio/menu widget eosremoterelease: 4 (Release Full) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Reading exposure... "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Setting new configuration OK. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Exposure complete. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] File added event completed. "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG] Downloading image... Name: (capt0000.cr2) Folder: (/) Delete from SD card? (false) "
org.kde.kstars.indi: Canon DSLR EOS 2000D : "[DEBUG]  Downloaded 0x0 (preview 0x0) "
org.kde.kstars.indi: Removing device "Canon DSLR EOS 2000D"
org.kde.kstars.indi: INDIListener: Removing device "Canon DSLR EOS 2000D"
org.kde.kstars.ekos: "Canon DSLR EOS 2000D is offline."
org.kde.kstars.indi: INDI Server:  "2020-07-30T12:20:12: Driver indi_canon_ccd: stderr EOF"
org.kde.kstars.indi: INDI Server:  "<delProperty device=\"Canon DSLR EOS 2000D\"/>"
org.kde.kstars.indi: INDI Server:  "Child process 144220 died"
org.kde.kstars.indi: INDI Server:  "2020-07-30T12:20:12: Driver indi_canon_ccd: Terminated after #0 restarts."
org.kde.kstars.indi: INDI Server:  ""
org.kde.kstars.indi: INDI driver  "indi_canon_ccd"  crashed!

knro, 2020-09-06:
I cannot reproduce this... can you run it via ekosdebugger? Select INDI tab and then run the driver and if it crashes share the backtrace.

BenjaminMerget, 2020-09-06:
Hi Jasem,
Sorry, I forgot to give an update here. I did a full and clean re-install of the latest Ubuntu and the issue resolved itself. I'm also no longer able to reproduce the error and it seems like it was a quite specific problem with my previous system setup. Unfortunately, I have no idea what the problem was as also multiple re-installations of Kstars, INDI and all associated libraries did not solve the problem. Again, sorry if I caused any unnecessary work.
Cheers,
Ben

closed 2020-09-06 by BenjaminMerget