indilib/indi-3rdparty #180: Queue Report pile up in indi_gpsd

rtlprmft, 2020-08-03:
I am not sure whether this is a general issue or depends on my configuration. I am using a ublox7-usb module with indi_gpsd. When I enable the automatic refresh (1s, 5s, or even manually), for every refresh, the time only increases by only one second. To my understanding, this is because the gpsd delivers a continuous stream and buffers this stream. When the buffer is then read with gps->read() only the next (but not the most recent report gets read. Therefore, I suggest the following change (works for me flewlessly):

Wrap the gps->read()  in w a while loop which makes sure that the queue is empty:

    while (1)
    {
        if ((gpsData = gps->read()) == nullptr)
        {
            LOG_ERROR("GPSD read error.");
            IDSetText(&GPSstatusTP, nullptr);
            return IPS_ALERT;
        }

        if (!gps->waiting(0))
            break;
    }

knro, 2020-08-03:
Thank you, I noticed this issue too. 

@rkaczorek What do you think?

rkaczorek, 2020-08-03:
Seems reasonable to me. @rtlprmft could you submit PR so I can review complete change?

jochym, 2020-08-04:
That is an obvious bug. I second the need for a fix. If you have a a patch for that @rtlprmft please submit a PR, otherwise we (authors of gpsd) need to fix this.

rtlprmft, 2020-08-04:
I am sorry, but I have no idea how to do so. Code snippet that solves the problem (empties the queue) is above. I am happy for any advice.

rkaczorek, 2020-08-04:
@jochym are you going to handle it?

jochym, 2020-08-05:
@rkaczorek I'll do it today

closed 2020-08-08 by knro