indilib/indi-3rdparty #181: indi_gpsd and system coordinates

rtlprmft, 2020-08-03:
indi_gpsd has a nice feature that is to provide the system time rather than the GPS time. This allows to use an identical configuration, although the GPS device is not available. It would be even nicer, if in case the device can not connect, a simulation mode could be enables in which the system time and a user supplied position is distributed.

knro, 2020-08-03:
Sure, I can see how this can be useful. Please submit a PR.

rtlprmft, 2020-08-04:
I have no idea how to do so... if I find time, I can maybe provide a diff.

jochym, 2020-08-05:
You can edit the file directly on github and commit your changes to a separate branch for review and merging. Even better you can fork the repo to your github account and edit it there. When you ready you can generate pull-request (PR) yourself or just signal us that your changes are ready. It is really easy. Actually easier then making a proper diff ;).

closed 2020-08-08 by knro