indilib/indi-3rdparty #183: Nikon 1 j4 with Indi

simont77, 2020-08-06:
Hi,

I tried to use my Nikon 1 J4 with ekos using indi_gphoto_ccd. I soon realized that if was not possible to set the shutterspeed and that the issue was with gphoto. I then contacted gphoto developer which was very efficient in supporting me. The latest commit of libgphoto now allows to set the shutterspeed on Nikon J4. I decided to try the new libgphoto with ekos, so I recompiled ekos on a Mac, using the script from rlancaste, modified by me to pick up the latest commit of libgphoto and not the stable release. Compiling was successful, but some issues still remains.

While I see the shutterspeed property under the capturesettings panel in Indi Control Panel, and I can indeed change the shutterspeed from there (before, the shutterspeed property does not even show up), when I try to change the shutterspeed in ekos, or try to use a Preset is the Main Control tab of Indi Control Panel, the capture is always done with the shutterspeed previously set in the camera (either by the camera own control, or using the capturesettings tabs). I disabled Force BULB (the camera does not support bulb mode).

Any idea of what's wrong?

Thanks

simont77, 2020-08-06:
Looking at the Indi DEBUG log, it seems that indi is trying to set property shutterspeed2 instead of shutterspeed.

```
GPhoto CCD : "[DEBUG] Finding optimal exposure setting for 0.25 seconds in shutterspeed2 (count=72)... "
GPhoto CCD : "[DEBUG] Closest match: 0.25 seconds Index: 24 "
GPhoto CCD : "[DEBUG] Setting radio/menu widget shutterspeed2: 24 (1/4) "
GPhoto CCD : "[DEBUG] Setting new configuration OK. "
GPhoto CCD : "[DEBUG] Using predefined exposure time: 0.25 seconds "
```

I'm not able to tell if this has to be fixed in gphoto or indi....
@msmeissn ?

knro, 2020-08-06:
So the driver check if shutterspeed2 exists first before checking for other shutterspeeds, since shutterspeed2 exists, it uses that.

msmeissn, 2020-08-06:
hmm, for the NIkon DSLR and Z the shutterspeed2 is the prefered one, but for 1 it would be shutterspeed ... 

I might just rename it again for Nikon 1 to match expectations

simont77, 2020-08-07:
Renaming worked, continuing discussion onto libghoto2 github page.

closed 2020-08-07 by simont77