indilib/indi-3rdparty #187: indi_gpsd incompatible with gpsd 3.21

mobarre, 2020-08-09:
Here is the build error:
```
[ 84%] Building CXX object indi-gpsd/CMakeFiles/indi_gpsd.dir/gps_driver.cpp.o
/home/mobarre/AUR/libindi_3rdparty/src/indi-3rdparty-1.8.5/indi-gpsd/gps_driver.cpp: In member function ‘virtual IPState GPSD::updateGPS()’:
/home/mobarre/AUR/libindi_3rdparty/src/indi-3rdparty-1.8.5/indi-gpsd/gps_driver.cpp:252:18: error: ‘struct gps_data_t’ has no member named ‘status’
  252 |     if (gpsData->status == STATUS_NO_FIX)
      |                  ^~~~~~
make[2]: *** [indi-gpsd/CMakeFiles/indi_gpsd.dir/build.make:82: indi-gpsd/CMakeFiles/indi_gpsd.dir/gps_driver.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2131: indi-gpsd/CMakeFiles/indi_gpsd.dir/all] Error 2
make: *** [Makefile:171: all] Error 2
``
The following comment has been left in gps.h:
      Move gps_data_t->status to gps_fix_t.status for better fix merging
       Add wspeedt, wspeedr, wanglem, wanglet, wangler to  gps_fix_t
       Remove unused gps_data_t.navadata_t.
       Add relPosL and relPosH to gps_fix_t.NED

Looking at their gitlab repo, it seems to have happened with release 3.21 of gpsd

Cheers !

knro, 2020-08-09:
Any #ifdef patch to fix this by checking the version?

knro, 2020-08-09:
@jochym What do you think?

mobarre, 2020-08-09:
#ifdef might work indeed !
in gps.h (from gpsd package) we have:
```
#define GPSD_API_MAJOR_VERSION  10      /* bump on incompatible changes */
#define GPSD_API_MINOR_VERSION  0       /* bump on compatible changes */
```

version 9.1 should have the former structure. 10.0 is the new one.

knro, 2020-08-09:
Great, can you please submit a PR to check for this?

mobarre, 2020-08-09:
Sure, with pleasure ! Let me get my shit together, I might be able to send it today

jochym, 2020-08-09:
I am afraid we will need to carry conditional code for some time. If you can submit the PR @mobarre , I can find some time to review this and we will get it fixed sooner. Otherwise it will need to wait at least a 5-10 days. I have a deadline coming ...

mobarre, 2020-08-09:
There you go :)

knro, 2020-08-13:
Looks like the issue is fixed. Please reopen if still persists.

closed 2020-08-13 by knro