indilib/indi-3rdparty #193: Driver indi_toupcam_ccd crash

gogolander, 2020-08-17:
Hi.
I'm trying to add a guide cam to my rig: I bought a Omegon Pro guide cam. I understand that this cam uses indi_toupcam_ccd driver but the the driver crashes at start:

    2020-08-17T12:18:57: Driver indi_toupcam_ccd: stderr EOF
    2020-08-17T12:18:57: Driver indi_toupcam_ccd: restart #9
    2020-08-17T12:18:57: Driver indi_toupcam_ccd: pid=1561 rfd=6 wfd=12 efd=14
    Child process 1556 died
    2020-08-17T12:18:58: Driver indi_altair_ccd: stderr EOF
    2020-08-17T12:18:58: Driver indi_altair_ccd: Terminated after #10 restarts.
    Child process 1559 died
    2020-08-17T12:18:58: Driver indi_toupcam_ccd: stderr EOF
    2020-08-17T12:18:58: Driver indi_toupcam_ccd: restart #10
    2020-08-17T12:18:58: Driver indi_toupcam_ccd: pid=1563 rfd=4 wfd=8 efd=12
    Child process 1561 died
    2020-08-17T12:18:59: Driver indi_toupcam_ccd: stderr EOF
    2020-08-17T12:18:59: Driver indi_toupcam_ccd: Terminated after #10 restarts.
    Child process 1563 died

**Desktop (please complete the following information):**
 - OS: latest Raspbian OS running on a Raspberry Pi 1B+
 - Indiserver: latest installed using astroberry apt repository;

I have no more details to give :(

Could anyone help me?

knro, 2020-08-20:
Did you build it from source? or PPA? Try to run it via gdb, but 99% it's a build issue.

knro, 2020-09-01:
Closing as it turned out to be hardware issue from what I heard.

closed 2020-09-01 by knro