indilib/indi-3rdparty #208: Pentax K-3 PTP support

liamstaras, 2020-08-29:
I have a Pentax K-3 that I've been experimenting with using the Karl's new driver. I previously had it working in some form with the gPhoto interface. It appears to work perfectly with the pkTriggerCord MSC interface (though I've not tested all features), though I'm wondering if it could run from the far more useful SDK PTP mode. Obviously this would enable features such as Live View, making this is genuinely plausible camera for astrophotography. I've had a look at the sources, and doing a simple search of the yucky binary blob reveals "K-3" listed with the other known supported cameras in multiple places. However, setting the camera to PTP and running the driver returns the "No supported Pentax cameras..." error. Again, I return to the sources and it would seem from some frivolous digging that the detector wrapper does not return a camera object for the K-3.
Normally, in this kind of situation I'd contact the development team for the SDK but I've _never_ had a response from ricoh.api. If anyone had any ideas for if this were possible I'd be very happy to test and/or debug the K-3's functionality on request!
Thanks is advance!

karlrees, 2020-09-02:
I've tried contacting Ricoh a few times, too, but never heard from them.  Figuring out how to make this work without their help is way above my pay grade--you'd probably have to start by reverse-engineering the SDK (which I think the licensing technically prohibits).  Of course, that would help out with a number of other problems we're having with the Ricoh libraries . . .

liamstaras, 2020-09-04:
Yes, it would seem like their support form feeds directly to /dev/null... Still, I'll keep trying to get through to them as I guess that (if queries actually do go somewhere useful) the more people that try, the more chance they'll finally respond. If only camera manufacturers would a) use the same protocol or b) open source their SDKs. Of course, they never will.
Cheers!

closed 2021-09-28 by knro