indilib/indi-3rdparty #210: Pentax Native Driver - wrong pixel size (Pentax K70)

noideaforaname, 2020-08-30:
While fetching information after taking the first picture the pixel size isn't correct. 

Size of a pixel is 3 instead 3.88. Leads to some issues while platesolving or photometric color correction in other programs.

[image]

[image]

knro, 2020-08-30:
@karlrees can you please check it out?

noideaforaname, 2020-09-01:
I tried to save the correct pixel values but they are overwritten with "3" when i restart the driver and take another picture.

karlrees, 2020-09-01:
I won't be able to look at it until the weekend, at least.  But it was supposed to work in the same way as the GPhoto driver, since I wasn't sure what to expect from the various camera models.  That is, you were supposed to supply the numbers the first time you used the driver with Ekos, and then it was saved after that.  At least, that's the way it's been working for me.  I certainly don't recall setting it to 3.0 anywhere in the code.  

Perhaps purging the driver configuration would let you reset it for now?

There should probably be a way I could fix the values for the K70 at least, so the user doesn't have to worry about it.  I'll look into that while I'm at it.

noideaforaname, 2020-09-01:
I did a little trial and error. I checked the files in /home/stellarmate/.indi

The values stored in the original and saved  "Pentax DSLR K70 (PTP Mode)_config" files are good. Pixel size is 3.88.

In "Pentax DSLR K70 (MFT Mode)_config.xml" nothing is seen about pixel size, frame size etc.... even when i save the driver configuration.

I tried to copy the values from the ptp file to the mft file but it doesn't help.

noideaforaname, 2020-09-03:
Checked again the reported values. The bitdepth is also wrong.

Correct values:

Max width: 6022
Max height: 4042
Pixel size: 3.88
Pixel size x: 3.88
Pixel size y: 3.88
Bit per Pixel: 14

karlrees, 2020-10-12:
The driver does not supply any of these values.  I didn't see an easy way to get them directly from the camera, and I didn't have time to go and research the values for each and every Pentax camera, so I instead opted for the same approach as GPhoto, which is to not set them at all.

My understanding is that if these values are not supplied by the driver, it's up to the client to supply them.  The first time you ever connected to the camera with Ekos, it should have asked you to supply the resolution and pixel size.  The numbers you supply are saved in the sqlite database, and then it appears that any time you connect the camera after that, Ekos sets these values for you.  

**The problem appears to be that there's no way in Ekos (that I can see) to reset the values after you specify the initial values.**   I suspect, if you're like me, the first time you started up client, you probably just wanted to get on with the process, and took a guess that the pixel size was 3 instead of 3.88 (or is it 3.89).  So now you're stuck with 3, unless you delete the sqlite database, or at least open it up and change the table by hand.  We should probably verify with @knro if that's the way this was designed.  If so, it would probably make sense to file an issue with Kstars/Ekos.  

What I'm not clear on is 1) why these values were set/saved in your PTP mode config file, but not your MSC mode config file; and 2) whether, if there's a way to save them in the MSC mode config file, would Ekos still have overridden them.  I'll have to look into that a bit more.

In any event, I'm going to treat this in part as an enhancement request for the driver to supply valid values for known cameras (i.e. the K70), and leave others unset (I think that should be possible).

knro, 2020-10-12:
In Ekos capture module, there is a trash icon next to the camera name to reset these value and it would popup a dialog again to enter them again.

noideaforaname, 2020-10-12:
When I enter the correct values after a fresh Install (from scratch) or delete them they are always set to 3 again after i restart or load the driver again.

Platesolving was working the last time with the value of 3... maybe it took a little longer.

When doing processing i can simply set the correct value in SIRIL while doing a color calibration.

knro, 2020-10-12:
Oh right the "3" issue was fixed by KStars in recent nightly builds. Please try that or wait for KStars 3.5.0 hopefully soon.

noideaforaname, 2020-10-12:
I'll wait a little bit, my camera died 2 weeks ago and is waiting to be serviced. Delay due corona.

closed 2021-05-25 by knro