indilib/indi-3rdparty #212: Trouble building indi-nexdome driver from git

xsnrg, 2020-09-02:
```
/usr/bin/c++   -I/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome -I/usr/include/libindi -I/usr/include/libnova  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g   -o CMakeFiles/indi_nexdome.dir/nex_dome.cpp.o -c /builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp: In member function 'virtual IPState NexDome::Park()':
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp:712:46: error: 'ShutterParkPolicySP' was not declared in this scope; did you mean 'ShutterFactorySP'?
  712 |     if (HasShutter() && IUFindOnSwitchIndex(&ShutterParkPolicySP) == SHUTTER_CLOSE_ON_PARK)
      |                                              ^~~~~~~~~~~~~~~~~~~
      |                                              ShutterFactorySP
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp:712:70: error: 'SHUTTER_CLOSE_ON_PARK' was not declared in this scope
  712 |     if (HasShutter() && IUFindOnSwitchIndex(&ShutterParkPolicySP) == SHUTTER_CLOSE_ON_PARK)
      |                                                                      ^~~~~~~~~~~~~~~~~~~~~
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp: In member function 'virtual IPState NexDome::UnPark()':
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp:729:46: error: 'ShutterParkPolicySP' was not declared in this scope; did you mean 'ShutterFactorySP'?
  729 |     if (HasShutter() && IUFindOnSwitchIndex(&ShutterParkPolicySP) == SHUTTER_OPEN_ON_UNPARK)
      |                                              ^~~~~~~~~~~~~~~~~~~
      |                                              ShutterFactorySP
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp:729:70: error: 'SHUTTER_OPEN_ON_UNPARK' was not declared in this scope
  729 |     if (HasShutter() && IUFindOnSwitchIndex(&ShutterParkPolicySP) == SHUTTER_OPEN_ON_UNPARK)
      |                                                                      ^~~~~~~~~~~~~~~~~~~~~~
/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome/nex_dome.cpp:743:1: warning: control reaches end of non-void function [-Wreturn-type]
  743 | }
      | ^
make[2]: *** [CMakeFiles/indi_nexdome.dir/build.make:86: CMakeFiles/indi_nexdome.dir/nex_dome.cpp.o] Error 1
make[2]: Leaving directory '/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome'
make[1]: Leaving directory '/builddir/build/BUILD/indi-3rdparty-master/indi-nexdome'
make[1]: *** [CMakeFiles/Makefile2:99: CMakeFiles/indi_nexdome.dir/all] Error 2
make: *** [Makefile:153: all] Error 2
```

**To Reproduce**
On Fedora, perform the normal build routine, cmake;make

**Expected behavior**
build the driver without errors

**Desktop (please complete the following information):**
 - OS: Fedora 31 and 32 x86_64
 - git

xsnrg, 2020-09-02:
https://copr.fedorainfracloud.org/coprs/xsnrg/indi-3rdparty-bleeding/package/indi-nexdome/

This started failing recently.

knro, 2020-09-03:
libindi needs to be updated first since it depends on that.

xsnrg, 2020-09-06:
rebuilding did seem to fix it. Thanks!

closed 2020-09-06 by xsnrg