indilib/indi-3rdparty #213: add github webhook to spawn Copr builds for indi-3rdparty

xsnrg, 2020-09-02:
Like we set up for indi, let us also set up for indi-3rdparty.

Please create the following `application/json webhook` in github in the indilib/indi-3rdparty repo.

` https://copr.fedorainfracloud.org/webhooks/github/33815/2883ba5d-bea5-41a8-ac1a-8f9182d49896/ `

knro, 2020-09-06:
Done!

closed 2020-09-06 by knro

xsnrg, 2020-09-09:
The push of 26b5f8ac22c9a577eebb1d8f4fadca546163d6cf should have triggered a build, but it did not. Did application/json get set on this webhook?

xsnrg, 2020-09-09:
Also, is there a way to re-open these @knro ?

knro, 2020-09-09:
Yes it was set to json. I updated it again just in case.

xsnrg, 2020-09-09:
https://copr.fedorainfracloud.org/webhooks/github/33815/2883ba5d-bea5-41a8-ac1a-8f9182d49896/  is indeed the right URL. I wonder why it did not fire. Will check again at next new hash.

xsnrg, 2020-09-09:
b1e6db4 did not trigger it either.  the indilib one works so nicely, not sure what is going on here.  Do the pages look different between the repos?

``` How to use it:

    In your GitHub project, go to Settings / Webhooks.
    Click on the Add webhook button.
    Fill in the Payload URL field with the url above.
    Select application/json as the content type.
    Click the Add webhook button.
```