indilib/indi-3rdparty #216: indi_rpicam - unable to get proper colour in the photographs

baryah, 2020-09-13:
with RPi-4-8GB+HQ Camera+Astroberry -> the clicked picture is a monochrome 'fits' file with bayer pattern BGGR (Header). When I debayer it in DSS, the output is a rather pink looking image. I chose Generic BGGR in the Raw/FITS DDP Settings.

lboclboc, 2020-09-14:
Can you please upload an example of your fits image and Ill have a look?

cnieswand, 2020-10-17:
Hi, same issue here. Almost same configuration RPi-4-4GB + HQ Camera + Astroberry. I have to set the BAYERPAT=BGGR in the fits header manually (in the INDI driver settings). But trying to debayer with Fitsworks fails: it does not produce any reasonable color for any BAYER matrix settings ...
Any progress on that topic?

baryah, 2020-10-17:
> Can you please upload an example of your fits image and Ill have a look?

https://drive.google.com/drive/folders/1ONh3VJKFUWhNRVF_gH7LUuhdIn8UmdYE?usp=sharing

Google Drive Link with fits file as well as the tiff files generated with PIPP using GBBR and RGGB for debayering along with the  logs of PIPP

cnieswand, 2020-10-17:
I do not have an astro photo yet ... just an image from my office ... which makes it very obvious.
Have a look at the histogram of your original fits file ... strange ...

lboclboc, 2020-10-17:
Thanks, your debayerd files for sure does not look right. When I open the raw fits-file in kstars it looks ok. Does it look alright for you in kstars (remember to turn on the auto debayer FITS-option in kstars).
I will do some testing with fitswork and pipp also.

cnieswand, 2020-10-17:
Oh yes ... 

> .... (remember to turn on the auto debayer FITS-option in kstars).

This flag is not set by default ... looks better now, but this does not give the correct result yet.
What I have to do is to open the fits file in Fitswork, debayer with BGGR, BUT with Rmul=2.00 and Bmul=2.00.
With these settings colors are perfect!

cnieswand, 2020-10-17:
I guess that green is simply not divided by 2 ... there are 2 green pixels in the matrix ...

cnieswand, 2020-10-17:
@baryah : treating your fits via fitswork the same way also gives a reasonable result ...

cnieswand, 2020-10-17:
@baryah : and using DSS: Set red scale and blue scale in the Raw/FITS Settings to 2.00 and you get a nice result too.

lboclboc, 2020-10-18:
Ok, great news. I suppose that if I just multiply blue and red with 2 in the driver all would be fine. I dont want to divide green with 2 since we would loose one bit then, and since its 16 data, the 12bit raw data will fit ok even if I multiply by 2.

lboclboc, 2020-10-18:
I really think after reading up on this that this is correct behavior. I did find some errors and possible instabilities in the code and I will submit that shortly. Let me know if you disagree.

cnieswand, 2020-10-19:
If you mean multiply B&R by 2 ... fine, yes ... that should do it ...

lboclboc, 2020-10-19:
Yes, ok great. We can close this ticket then.

cnieswand, 2020-10-19:
From my side, yes.
Thank you very much for your help!
@baryah ???

baryah, 2020-10-19:
Sure. 
I believe progress has been made. Hope to get better results ..

closed 2021-09-28 by knro