indilib/indi-3rdparty #220: Error installation indi_rpicam

HamHamFonFon, 2020-09-26:
**Describe the bug**
Hello, i have an error on installation of driver indi_rpicam. Error message : 
```
CMake Error at /home/hamham/Projects/indi-3rdparty/cmake_modules/FindINDI.cmake:285 (message):
  Could not find INDI include directory
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hamham/Projects/build/indi-rpicam/CMakeFiles/CMakeOutput.log"
```

**To Reproduce**
Here all steps : 
First, installation of indi many days ago from synaptic, but i've try again just to check installed version : 
```
hamham@astroloutre:~$ sudo apt-get install indi-full
indi-full is already latest version (1.8.6~202008211454~ubuntu18.04.1).
hamham@astroloutre:~$ sudo apt-get install indi-full kstars-bleeding
indi-full is already latest version (1.8.6~202008211454~ubuntu18.04.1).
kstars-bleeding is already latest version (6:3.5.0+202008282324~ubuntu18.04.1).
Following packages have been automatically installed and will be removed  :
  libqt5multimedia5 libqt5multimediawidgets5 libqt5opengl5 libqt5positioning5 libqt5serialport5
hamham@astroloutre:~$ sudo add-apt-repository ppa:mutlaqja/ppa
hamham@astroloutre:~$ sudo apt-get update
hamham@astroloutre:~$ sudo apt-get install indi-webcam
//installation ok...
```
Then, installation and clone of indi-3rdparty
```
hamham@astroloutre:~$ sudo apt-get -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libdc1394-22-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev

libraw-dev is already latest version (0.18.8-1ubuntu0.3).
libjpeg-dev is already latest version (1:1.5.2-2~bionic1.0).
Following packages will be installed : 
  cmake-data dpkg-dev fakeroot g++ g++-7 gir1.2-harfbuzz-0.0 icu-devtools libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl libavdevice58 libavfilter-dev libavfilter7 libavformat-dev libavutil-dev
  libboost-regex1.65-dev libboost-regex1.65.1 libboost1.65-dev libcfitsio-doc libexif-dev libexif-doc libfakeroot libfftw3-bin libftdi1-doc libglib2.0-dev libglib2.0-dev-bin libgraphite2-dev libharfbuzz-dev libharfbuzz-gobject0
  libicu-dev libicu-le-hb-dev libicu-le-hb0 libiculx60 libjbig-dev liblilv-0-0 liblimesuite17.12-1 liblzma-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpostproc-dev libraw1394-dev libraw1394-tools librhash0 librtlsdr0
  libserd-0-0 libsord-0-0 libsratom-0-0 libstdc++-7-dev libswresample-dev libswscale-dev libtiff5-dev libusb-1.0-doc libusb-dev libuv1 libvidstab1.1 limesuite-udev make pkg-config
Suggered packages :
  cmake-doc ninja-build debian-keyring g++-multilib g++-7-multilib gcc-7-doc libstdc++6-7-dbg libboost-doc libboost1.65-doc libboost-atomic1.65-dev libboost-chrono1.65-dev libboost-container1.65-dev libboost-context1.65-dev
  libboost-coroutine1.65-dev libboost-date-time1.65-dev libboost-exception1.65-dev libboost-fiber1.65-dev libboost-filesystem1.65-dev libboost-graph1.65-dev libboost-graph-parallel1.65-dev libboost-iostreams1.65-dev
  libboost-locale1.65-dev libboost-log1.65-dev libboost-math1.65-dev libboost-mpi1.65-dev libboost-mpi-python1.65-dev libboost-numpy1.65-dev libboost-program-options1.65-dev libboost-python1.65-dev libboost-random1.65-dev
  libboost-serialization1.65-dev libboost-signals1.65-dev libboost-stacktrace1.65-dev libboost-system1.65-dev libboost-test1.65-dev libboost-thread1.65-dev libboost-timer1.65-dev libboost-type-erasure1.65-dev libboost-wave1.65-dev
  libboost1.65-tools-dev libmpfrc++-dev libntl-dev libcurl4-doc libgnutls28-dev libidn11-dev libkrb5-dev libldap2-dev librtmp-dev libssh2-1-dev libfftw3-doc libglib2.0-doc libgraphite2-utils icu-doc liblimesuite-doc liblzma-doc
  libraw1394-doc serdi sordi libstdc++-7-doc make-doc
NEW packages will be installed :
  build-essential cmake cmake-data dpkg-dev fakeroot g++ g++-7 gir1.2-harfbuzz-0.0 icu-devtools libalgorithm-diff-perl libalgorithm-diff-xs-perl libalgorithm-merge-perl libavcodec-dev libavdevice-dev libavdevice58 libavfilter-dev
  libavfilter7 libavformat-dev libavutil-dev libboost-dev libboost-regex-dev libboost-regex1.65-dev libboost-regex1.65.1 libboost1.65-dev libcfitsio-dev libcfitsio-doc libcurl4-gnutls-dev libdc1394-22-dev libexif-dev libexif-doc
  libfakeroot libfftw3-bin libfftw3-dev libftdi-dev libftdi1-dev libftdi1-doc libglib2.0-dev libglib2.0-dev-bin libgphoto2-dev libgps-dev libgraphite2-dev libgsl-dev libharfbuzz-dev libharfbuzz-gobject0 libicu-dev libicu-le-hb-dev
  libicu-le-hb0 libiculx60 libjbig-dev liblilv-0-0 liblimesuite-dev liblimesuite17.12-1 liblzma-dev libnova-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpostproc-dev libraw1394-dev libraw1394-tools librhash0 librtlsdr-dev
  librtlsdr0 libserd-0-0 libsord-0-0 libsratom-0-0 libstdc++-7-dev libswresample-dev libswscale-dev libtiff-dev libtiff5-dev libusb-1.0-0-dev libusb-1.0-doc libusb-dev libuv1 libvidstab1.1 limesuite-udev make pkg-config zlib1g-dev

hamham@astroloutre:~$ mkdir -p ~/Projects
hamham@astroloutre:~$ cd Projects/
hamham@astroloutre:~/Projects$ git clone --depth=1 https://github.com/indilib/indi-3rdparty
hamham@astroloutre:~/Projects$ mkdir -p build/indi-rpicam
hamham@astroloutre:~/Projects$ cd build/indi-rpicam/
hamham@astroloutre:~/Projects/build/indi-rpicam$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
CMake Error at /home/hamham/Projects/indi-3rdparty/cmake_modules/FindINDI.cmake:285 (message):
  Could not find INDI include directory
Call Stack (most recent call first):
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hamham/Projects/build/indi-rpicam/CMakeFiles/CMakeOutput.log".
```

**Desktop (please complete the following information):**
 - Raspberry Py 3 Model B+
Version of ubuntu : 
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```
Architecture : 
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ uname -a
Linux astroloutre 4.15.0-1067-raspi2 #71-Ubuntu SMP PREEMPT Fri Jul 10 01:37:51 UTC 2020 armv7l armv7l armv7l GNU/Linux
```

Thank you :)

lboclboc, 2020-09-28:
Hi, you also need libindi-dev installed in order to be able to build it.
Regards, Lars

HamHamFonFon, 2020-09-28:
All right, i have install libindi-dev.
Command 
```
 cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
```
is OK : 
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
-- Found INDI: /usr/lib/arm-linux-gnueabihf/libindidriver.so;/usr/lib/arm-linux-gnueabihf/libindiAlignmentDriver.so (found version "1.8.6") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hamham/Projects/build/indi-rpicam
```

But commande make -j4 return some errors : 
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ make -j4
Scanning dependencies of target indi_rpicam
[  7%] Building CXX object CMakeFiles/indi_rpicam.dir/indi_rpicam.cpp.o
[ 15%] Building CXX object CMakeFiles/indi_rpicam.dir/mmaldriver.cpp.o
[ 23%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalcamera.cpp.o
[ 30%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalencoder.cpp.o
/home/hamham/Projects/indi-3rdparty/indi-rpicam/mmalencoder.cpp:22:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
CMakeFiles/indi_rpicam.dir/build.make:134: recipe for target 'CMakeFiles/indi_rpicam.dir/mmalencoder.cpp.o' failed
make[2]: *** [CMakeFiles/indi_rpicam.dir/mmalencoder.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/hamham/Projects/indi-3rdparty/indi-rpicam/mmalcamera.cpp:22:10: fatal error: mmal_logging.h: No such file or directory
 #include <mmal_logging.h>
          ^~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/indi_rpicam.dir/build.make:62: recipe for target 'CMakeFiles/indi_rpicam.dir/mmalcamera.cpp.o' failed
make[2]: *** [CMakeFiles/indi_rpicam.dir/mmalcamera.cpp.o] Error 1
In file included from /home/hamham/Projects/indi-3rdparty/indi-rpicam/mmaldriver.h:26:0,
                 from /home/hamham/Projects/indi-3rdparty/indi-rpicam/indi_rpicam.cpp:24:
/home/hamham/Projects/indi-3rdparty/indi-rpicam/cameracontrol.h:25:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
In file included from /home/hamham/Projects/indi-3rdparty/indi-rpicam/mmaldriver.h:26:0,
                 from /home/hamham/Projects/indi-3rdparty/indi-rpicam/mmaldriver.cpp:33:
/home/hamham/Projects/indi-3rdparty/indi-rpicam/cameracontrol.h:25:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
CMakeFiles/indi_rpicam.dir/build.make:86: recipe for target 'CMakeFiles/indi_rpicam.dir/indi_rpicam.cpp.o' failed
make[2]: *** [CMakeFiles/indi_rpicam.dir/indi_rpicam.cpp.o] Error 1
CMakeFiles/indi_rpicam.dir/build.make:110: recipe for target 'CMakeFiles/indi_rpicam.dir/mmaldriver.cpp.o' failed
make[2]: *** [CMakeFiles/indi_rpicam.dir/mmaldriver.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/indi_rpicam.dir/all' failed
make[1]: *** [CMakeFiles/indi_rpicam.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

```

lboclboc, 2020-09-28:
You need to run the compilation on a raspberry. Im not sure if crosscompilation is available for libindi.
Ill make sure this information is added to the README.
Regards, Lars

HamHamFonFon, 2020-09-28:
I'm already on my raspberry pi :). I've mentionned it on my first message : 
Raspberry Pi 3 Model B+ / Ubuntu 18.04.5 LTS, not raspbian.

lboclboc, 2020-09-28:
Hm, ok. Normally the mmal stuff is installed under /opt/vc, maybe the Ubuntu dist installs it under some other place. Do a dpkg - L raspberrypi-deb, you should see it there. You should then be able to add -DCMAKE_LIBRARY_PATH=path -DCMAKE_INCLUDE_PATH=path to cmake

HamHamFonFon, 2020-09-28:
Sorry, it doen't exist :
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ sudo dpkg -L raspberrypi-deb
dpkg-query: package « raspberrypi-deb » is not installed
Use dpkg --info (= dpkg-deb --info) to examine archive files, and dpkg --contents (= dpkg-deb --contents) to show their leur content.
```

lboclboc, 2020-09-29:
Sorry, i meant dpkg -L rasberrpi-dev (typo on my mobile)

HamHamFonFon, 2020-09-29:
I have pulled the pull request https://github.com/indilib/indi-3rdparty/pull/221 and all have been worked : 
```
hamham@astroloutre:~/Projects/build/indi-rpicam$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-rpicam
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Found INDI: /usr/lib/arm-linux-gnueabihf/libindidriver.so;/usr/lib/arm-linux-gnueabihf/libindiAlignmentDriver.so (found version "1.8.6") 
-- Found CFITSIO 3.43: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hamham/Projects/build/indi-rpicam
hamham@astroloutre:~/Projects/build/indi-rpicam$ make -j4
Scanning dependencies of target indi_rpicam
[  7%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalcamera.cpp.o
[ 21%] Building CXX object CMakeFiles/indi_rpicam.dir/indi_rpicam.cpp.o
[ 21%] Building CXX object CMakeFiles/indi_rpicam.dir/mmaldriver.cpp.o
[ 28%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalencoder.cpp.o
[ 35%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalexception.cpp.o
[ 42%] Building CXX object CMakeFiles/indi_rpicam.dir/mmalcomponent.cpp.o
[ 50%] Building CXX object CMakeFiles/indi_rpicam.dir/cameracontrol.cpp.o
[ 57%] Building CXX object CMakeFiles/indi_rpicam.dir/raw10tobayer16pipeline.cpp.o
[ 64%] Building CXX object CMakeFiles/indi_rpicam.dir/raw12tobayer16pipeline.cpp.o
[ 71%] Building CXX object CMakeFiles/indi_rpicam.dir/pipeline.cpp.o
[ 78%] Building CXX object CMakeFiles/indi_rpicam.dir/jpegpipeline.cpp.o
[ 85%] Building CXX object CMakeFiles/indi_rpicam.dir/broadcompipeline.cpp.o
[ 92%] Building CXX object CMakeFiles/indi_rpicam.dir/pipetee.cpp.o
[100%] Linking CXX executable indi_rpicam
[100%] Built target indi_rpicam
hamham@astroloutre:~/Projects/build/indi-rpicam$ sudo make install
[sudo] password for hamham: 
[100%] Built target indi_rpicam
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/bin/indi_rpicam
-- Set runtime path of "/usr/bin/indi_rpicam" to ""
-- Installing: /usr/share/indi/indi_rpicam.xml
hamham@astroloutre:~/Projects/build/indi-rpicam$ ll /usr/share/indi | grep rpicam
-rw-r--r--   1 root root   230 sept. 29 08:59 indi_rpicam.xml
```
It looks like fine :). 
I think we can close this issue and if have some problems, i will mention it.
Thank you

lboclboc, 2020-09-29:
Great, thanks.

closed 2020-09-29 by HamHamFonFon