indilib/indi-3rdparty #228: Trouble with AltaU apogee and libapogee

srswinde, 2020-10-22:
**The Bug**
I am using the indi_apogee_ccd driver to communicate with [this AltaU-42 camera](https://www.cfa.harvard.edu/mmti/wfs/mmtcam.html). The Client is KStars. Everything seems to work until I click the **disconnect** button. The camera disconnects and the driver gives the INDI message ```Message [INFO] Camera is offline. 2020-10-20T18:06:38```. I can then reconnect using the Connect button and all seems fine. The trouble comes when I attempt to take an exposure. I get an error message:
 ```Message [ERROR] StartExposure() failed. libapogee:/build/libapogee3-4BddDT/libapogee3-3.2+~202008211633~ubuntu18.04.1/libapogee/Alta.cpp(278):Invalid image status, 1, for starting an exposure:4. 2020-10-20T18:07:36 ```.

This issue also occurs if I simple control-c the indidriver and restart without hitting disconnect.

The only fix I can find so far is to power cycle the camera. If I do this and reconnect, the first exposure I take is messed up. The bottom bit of the image is saturated while the top looks normal. See image under screenshots. All images after the initial image look fine.

I went down the rabbit hole a bit to see if I can find the cause. The line that the INDI message is complaining about, is in the [Alta.cpp module ](https://github.com/indilib/indi-3rdparty/blob/master/libapogee/Alta.cpp#L276) in the StartExposure method. It is unhappy with the Camera state of 1, it should be 4. According to [CameraInfo.h](https://github.com/indilib/indi-3rdparty/blob/master/libapogee/CameraInfo.h), status of 1 is Status_Exposing while 4 is Status_Flushing. It seems the driver is waiting for the camera to start flushing, but it never does.

The [Disconnect method in the apogee driver](https://github.com/indilib/indi-3rdparty/blob/master/indi-apogee/apogee_ccd.cpp#L1240) really only calls the CloseConnection method on the ApgCam instance, which eventually gets us to [ApogeeCam::DefaultCloseConnection](https://github.com/indilib/indi-3rdparty/blob/master/libapogee/ApogeeCam.cpp#L289) in the random factory driver. This method calls CancelExposureNoThrow and then reset on a bunch of objects. It must be somewhere in here that the firmware on the camera gets into a funky state.

Separately, I compiled and ran the randomfactory libapogee SimpleExposure program. It seems to cause the same problem, so the issues is either with the camera or with libapogee. I will send randomfactory an email to see if they have any ideas. 

**To Reproduce**

1. Run the indi_apogee_ccd driver
2. Use KStars to connect with the Driver
3. Disconnect the driver
4. Connect to the driver again
5. Take an exposure and receive the above error.

**Expected behavior**
I would expect to be able to disconnect the driver from the camera without having to cycle power on the camera.

**Screenshots**

[image]

**Desktop**
 - indiserver and the driver run on Ubuntu 18.04:
 - The INDI client runs on windows 10.
 - This issue occurs whether I use the ubuntu binaries from **ppa:mutlaqja/ppa** or when I build it from the GIT master branch.

knro, 2020-10-23:
Thank you for the report. I don't think RandomFacotry maintains this anymore last time I checked, so it's better to check with Andor. The SDK is no longer updated since we don't have any upstream source to update against. If Andor has a multi-platform SDK we can use then it's great.

srswinde, 2020-10-27:
I heard back from Dave Mills at random factory. He gave me some suggestions involving calling cam.Reset() before disconnecting. It works some of the time . It seems as though this problem is specific to my camera. Unless anyone has any other thoughts, I will close this issue. I can work around it by cycling power on the camera whenever I disconnect from it. 

To your point Jasem, RandomFactory is not maintaining this software anymore though they seem interested in helping where they can. Andor has an SDK that they use for all their cameras, but it isn't open source. They stopped making Apogee Altas back in March, so I don't know if their SDK would even work on these cameras.

closed 2020-10-28 by srswinde