indilib/indi-3rdparty #230: indi-gphoto indi_fuji_ccd Fuji X-T2 Shutter not activated after the 18th image in a sequence

tvaucher, 2020-10-27:
**Describe the bugs**
When using a Fuji X-T2 with the sequencer from Kstars/Ekos and `FORCE BULB` off (so using the internal shutter speed), the sequence goes well for a while (18 images), but then the shutter is not activated anymore and stale data is downloaded (see `FujiX-T2-T-test.log`) but the sequence doesn't fail and no errors are reported. I though it could be related to #151 but the patch appears to have been pushed on the main branch.

When trying to do the same thing with `FORCE BULB` on, the sequences goes well for a while (18 images) but then the shutter can't be activated and the sequence fail (see `FujiX-T2-Bulb-test.log`). Exactly as described [here](https://github.com/gphoto/libgphoto2/issues/506#issuecomment-709592944).

In both case, this doesn't seem to be related to exposure duration (tested at 120sec, 5sec and 1sec), nor the length of the sequence (tested at 20, 30, 40 images) and it always fail after the same number of exposures. The fact that it always fail after the same number of exposure makes me think the origin of the issue if in `indi-gphoto` rather than in `libgphoto2`.

**To Reproduce**

1. Make Install latest version of `libgphoto2`, `gphoto2` (both 2.5.26.1) and `indi-gphoto`
2. In Kstars/Ekos, connect the camera and select `FORCE BULB` on or off depending on which issue to reproduce
3. Create a sequence in the imaging tab with more than 18 images
4. Start the sequence
5. After 18 images, observe the error mentioned in the description

**Expected behavior**
A sequence should be able to successfully capture an unbounded number of images. 

**Desktop (please complete the following information):**
 - Hardware: Raspberry Pi 4, 2Go RAM
 - OS: Astroberry Server 2.0.3
 - Version: Additionally, libgphoto2 and gphoto2 where make installed from github to get the latest versions (2.25.6.1) and relinked. The indi-gphoto2 driver was remake installed in order to get the latest commit that handle the particularities with shutter duration in Fuji cameras.

**Log Files**
1.  [FujiXT2-T-test.log](https://github.com/indilib/indi-3rdparty/files/5448632/FujiXT2-T-test.log)
2. [FujiXT2-Bulb-test.log](https://github.com/indilib/indi-3rdparty/files/5448631/FujiXT2-Bulb-test.log)

Thanks in advance for looking into it, any support and advancement would be a big plus towards the adoption of Fuji cameras for DSO imaging!

tvaucher, 2020-11-01:
Quick update.

After a bit of thinking, I guess we all agree that 18 was really a odd number, and reading a few hundreds (thousands?) lines of C++ I couldn't find something that could cause the issue in the code.

So I then looked at the camera and something that caught my attention was the status led. Basically when I was taking normal test captures / images for the plate solver, the capture would happen, then the led status would blink green/orange while downloading (normal in Fuji cameras while the image is being written to the SD card too), then turn off. But when taking a sequence, the led status never turned off and keep blinking green/orange thus being constantly "busy" writing. After the 18th capture, the led would actually turn solid orange (took me a while to catch that), which means that no new image can be taken as it's saving the images. The difference between the 2 executions? Images are being written to the SD card. So I set "SD Image" to delete and tried to take a new sequence ... and _voilà_, no issues anymore, the status led shows the same pattern as a normal execution (i.e. turns off between the exposure).

**TL;DR**: Trying to take a sequence with `"SD Image"  = Save` appears to fill the buffer of the camera, preventing any image to be captured after 18 images (i.e. imo, when the buffer is full). Setting  `"SD Image"  = Delete` appears to solve the issue (at the expense of only having no local copy of your images). Maybe a note somewhere in the would be interesting for future Fuji user 😄 

PS: Also adding that to use Fuji Bulb and the right Fuji shutter speed you need `libgphoto2 >= 2.5.26` would be valuable!

m-kostrzewa, 2020-12-17:
>  I though it could be related to #151 but the patch appears to have been pushed on the main branch.

Where? Git blame shows latest modification of gphoto_read_exposure_fd function 2 years ago

daveswin, 2021-02-06:
I'm glad I found this. The hint "SD Image" = Delete solved the issue I had with large sequences being sometimes corrupt. Corrupt in the sense that eventually all the saved files are the same file just renamed according to the defined naming rules, or having the wrong exposure time of ISO set.  Previously I had noticed no difference between "SD Image" = Delete and Save - the image was saved (according to the naming rule of the camera - Fuji X-T3 in my case).

closed 2021-05-25 by knro