indilib/indi-3rdparty #232: build problem with indi-ahp-correlator

xsnrg, 2020-11-03:
The two most recent builds of indi-ahp-correlator have failed with the problem:

```
[ 50%] Building CXX object CMakeFiles/indi_ahp_correlator.dir/indi_ahp_correlator.cpp.o
/usr/bin/g++  -I/builddir/build/BUILD/indi-ahp-correlator-1.8.7.git/indi-ahp-correlator -I/usr/include/libindi -I/usr/include/cfitsio -I/usr/include/ahp -O2  -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -std=c++11 -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -o CMakeFiles/indi_ahp_correlator.dir/indi_ahp_correlator.cpp.o -c /builddir/build/BUILD/indi-ahp-correlator-1.8.7.git/indi-ahp-correlator/indi_ahp_correlator.cpp
In file included from /builddir/build/BUILD/indi-ahp-correlator-1.8.7.git/indi-ahp-correlator/indi_ahp_correlator.cpp:28:
/builddir/build/BUILD/indi-ahp-correlator-1.8.7.git/indi-ahp-correlator/indi_ahp_correlator.h: In destructor 'virtual Interferometer::~Interferometer()':
/builddir/build/BUILD/indi-ahp-correlator-1.8.7.git/indi-ahp-correlator/indi_ahp_correlator.h:48:28: error: 'xc_get_nbaselines' was not declared in this scope; did you mean 'ahp_xc_get_nbaselines'?
   48 |         for(int x = 0; x < xc_get_nbaselines(); x++)
      |                            ^~~~~~~~~~~~~~~~~
      |                            ahp_xc_get_nbaselines
```

more details and similar errors here:

https://download.copr.fedorainfracloud.org/results/xsnrg/indi-3rdparty-bleeding/fedora-33-x86_64/01743657-indi-ahp-correlator/builder-live.log.gz

I have also verified the library (libahp_xc) is current master.

xsnrg, 2020-11-03:
it appears to be failing because of changes made Nov 1 to the library header, with there being no recent changes to the driver. @iliaplatone

iliaplatone, 2020-11-03:
Thanks, sorry this was because a misalignment between my local repos

closed 2020-11-13 by xsnrg