indilib/indi-3rdparty #233: Read Mode - disappeared from the INDI GUI menu

JBielanski, 2020-11-12:
After update to new INDI version from git, Read Mode disappeared from menu.
Number of modes for QHY600 read properly. Following things does not working:
- indigui do not display 'Read Mode', only in logs is displayed maximum number of read modes and current read mode
- read mode is not read from camera config file

knro, 2020-11-12:
Nothing recently changed AFAIK. ReadModeNP is defined once HasReadMode is set to true. It is only set to true if number of read modes > 1. How many read modes does your camera have?

This code is doing this check:
```
ret = GetQHYCCDNumberOfReadModes(m_CameraHandle, &readModes);
if (ret == QHYCCD_SUCCESS && readModes > 1)
{
 HasReadMode = true;
 LOGF_INFO("Number of read modes: %zu", readModes);
}
```

JBielanski, 2020-11-13:
INDI has raported 4 read modes, but on not display 'read mode':
[image]
`11:31:06.357: Connecting to INDI server
11:31:06.759: Server connected
11:31:09.276: CONNECTION CONNECT=ON
11:31:13.633: [INFO] Connected to QHY600M-a7ca3c438f7ea8e27.
11:31:13.633: [INFO] Using QHY SDK version 20.9.30
11:31:13.633: [INFO] Number of read modes: 4
11:31:13.633: [INFO] Camera exposure limits: Min: 0.000001s Max: 3600s Step 0s
11:31:13.946: [INFO] Current read mode: 0
11:31:13.946: [INFO] Gain Settings: Value: 30 Min: 0 Max: 200 Step 1
11:31:13.980: [INFO] Offset Settings: Value: 30 Min: 0 Max: 255 Step 1
11:31:13.980: [INFO] USB Traffic Settings: Value: 50 Min: 0 Max: 60 Step 3`
On evening I will check the code. I am planning to find commit where everything works well and from this point I will start looking for commit where problem appear.

commit 12a22fea (2020-11-13): define readmode. Fixes #233
closed 2020-11-13 by knro