indilib/indi-3rdparty #237: aarch64

vedrafal, 2020-11-15:
Hello,
would it be possible to add/build aarch64 drivers for indi-3rdparty. 
I tried to build from source and it failed, looking into the folders I saw that arm64 and aarch64 are missing on libasi.
I'm only looking for the zwo and canon support. I'm not sure how much work that is and if it is even possible.

Greeting, Vedrafal

knro, 2020-11-15:
libasi is already available for arm64 and gphoto as well. What failure did you have exactly?

vedrafal, 2020-11-16:
I was trying to make indi-asi and indi-gphoto, during the process I was getting errors that I need libasi. 

Trying to to build libasi I got an error that libasi is not available for aarch64.
I'm not sure if I was just being an idiot but what I just did was add an aarch64 and arm64 folder into the libasi folder and copy the fles from the x64 into that on my pc then I managed to successfully build libasi and after that indi-asi and indi-gphoto compiled without any problems. 

I did not test the camera yet but kstars shows both the zwo and the canon driver in ekos now.

I am using Manjaro on raspberry pi 4 and the Aur package for indi-3rdparty was also not working claiming that it's not available for aarch64. So my thought was to just build the two drives myself.

closed 2020-11-20 by vedrafal