indilib/indi-3rdparty #238: libahp_xc doesn't install and indi-ahp-correlator fail to build

nouspiro, 2020-11-18:
I am building latest indilib on my raspberry pi. Last week it started to fail building indi-ahp-correlator on undefined functions.
Turn out that ahp_xc.h that was installed in /usr/include/ahp was old version. Rebuilding didn't updated this header file. I must add these lines into IF(BUILD_LIBS) section so libahp_xc get installed during lib building phase and then indi-ahp-correlator can build correctly

`#if (WITH_AHP_CORRELATOR)`
`add_subdirector(libahp_xc)`
`endif(WITH_AHP_CORRELATOR)`

knro, 2020-11-18:
@iliaplatone Can you please check?

iliaplatone, 2020-11-18:
I made a pull request, it now stops at astrolink build

iliaplatone, 2020-11-18:
The library was not being installed in CircleCI
PR is #240

nouspiro, 2020-11-20:
That merge request doesn't fix issue. It still doesn't build/install libahp_xc with BUILD_LIBS=1. Instead it get build/installed during normal driver build but then it still doesn't build indi-ahp-correlator

commit 9c2e9832 (2020-11-20): Add libahp_xc to BUILD_LIBS which should fix #238
closed 2020-11-20 by knro