indilib/indi-3rdparty #250: Celestron AUX errors

knro, 2020-11-29:
Driver on MacOS fails to communicate properly with Celestron Nexstar Evolution via WiFi.

Complete log is attached.

Initial handshake appears to be going fine, but then it starts failing to parse packets:
```
[2020-11-28T18:09:33.899 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. char 4 is 49 ascii I "
[2020-11-28T18:09:33.901 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Receive packet: 40 FE B6 EA FE 3B 03 0E "
[2020-11-28T18:09:33.954 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. char 4 is 49 ascii I "
[2020-11-28T18:09:33.957 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Receive packet: 40 FE B6 EA FE 01 DE 3B "
[2020-11-28T18:09:35.010 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. char 4 is 49 ascii I "
[2020-11-28T18:09:35.012 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Receive packet: 40 FE B6 EA FE 10 0E 01 "
[2020-11-28T18:09:35.074 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. char 4 is 49 ascii I "
[2020-11-28T18:09:35.075 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Receive packet: 40 FE B6 EA FE 64 5D 4D "
[2020-11-28T18:09:36.117 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. char 4 is 49 ascii I "
[2020-11-28T18:09:36.119 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Receive packet: 40 FE B6 EA FE 03 0E 11 "
[2020-11-28T18:09:36.172 EST INFO ][           org.kde.kstars.indi] - Celestron WiFi :  "[ERROR] Resp. 
```
[Logs.zip](https://github.com/indilib/indi-3rdparty/files/5611783/Logs.zip)

knro, 2020-11-29:
@fabriziop @jochym Any ideas on what might be causing this? Any requirements for minimum firmware for example? I recall connecting to another evolution mount a couple of months ago without any issues.

knro, 2020-11-29:
@fabriziop @jochym completely unrelated, but there is a thread on INDI forum that involves this driver and perhaps requires your input (I have very little knowledge of the details).

fabriziop, 2020-11-29:
> @fabriziop @jochym Any ideas on what might be causing this? Any requirements for minimum firmware for example? I recall connecting to another evolution mount a couple of months ago without any issues.

I looked at the code: that error may happen for serial connection to HC only. So, it seems that the driver failed to detect the WiFi connection. Since I have only wired connections, that part was developed by Jochym. If you can give the connection log of Kstars INDI control panel of the driver, perhaps I can go deeper.

fabriziop, 2020-11-29:
> @fabriziop @jochym completely unrelated, but there is a thread on INDI forum that involves this driver and perhaps requires your input (I have very little knowledge of the details).
> 
> https://indilib.org/forum/development/8246-indi-celestron-aux.html

Thanks for the information. I'll contribute to the thread.

closed 2021-09-28 by knro