indilib/indi-3rdparty #252: Please add fxload as depedent for libqhy

JiangXL, 2020-12-07:
**Describe the bug**
Without `fxload`, `libqhy` can not run `RUN+="/sbin/fxload -t fx3 -I /lib/firmware/qhy/QHYXXX.img -D $env{DEVNAME}` inside `85-qhyccd.rules`. As a result, indi show that no camera is found.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Install libqhy & ind-qhy from copr or aur
2. run indi or qhy_ccd_test
3. show error, "no camera found"

**Expected behavior**
indi can recognize qhyccd camera 

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Fedora 33 and Manjaro
 - Version: master

knro, 2020-12-07:
@xsnrg Is this a Fedora spec issue?

xsnrg, 2020-12-07:
@knro should be fixed by #253 , though I do not have a QHY camera to test with. @JiangXL , please do a `dnf update --refresh` to pull the latest package 15 or so minutes after this is merged.

PR #253 (2020-12-07): fxload_spec: add the Require: fxload entry for libqhy
fixes #252
closed 2020-12-07 by knro