indilib/indi-3rdparty #257: Another `fxload ` which has `-fx3` option is required to load qhy usb3 camera

JiangXL, 2020-12-11:
**Describe the bug**
Although #252  enable my QHY5L-ll-M, but QHY163M is still fail to detect. I found that fedora's fxload hasn't `fx3` to load usb3 firmware by `RUN+="/sbin/fxload -t fx3 -I /lib/firmware/qhy/QHY163.img` in rules  file. QHY5L-ll-M only require `fx2` option.

After searching at network, I find [this reply from qhy](https://github.com/qhyccd-lzr/QHYCCD_Linux/issues/29).

Is it possible that build a new `fxload` in other copr or add `fxload` code inside `libqhy`?

Thanks.

**To Reproduce**
1. Install libqhy, indi-qhy(don't use qhyccd sdk own installation script )
2. Connect QHY163M(or other usb3 camera)
3. Run indi

**Expected behavior**
 Indi show that no camera was found.

**Desktop (please complete the following information):**
 - OS: Fedora 33
 - Version: from fedora copr

JiangXL, 2020-12-11:
USB3 compatible `fxload` can be found in https://github.com/libusb/libusb/blob/master/examples/fxload.c

xsnrg, 2020-12-11:
The best course of action would be to open a request against Fedora upstream. In the short term I can look at building it in another copr, but it may be some days yet.

JiangXL, 2020-12-11:
Thanks for your attention : )

xsnrg, 2020-12-12:
https://bugzilla.redhat.com/show_bug.cgi?id=1907078

knro, 2021-09-28:
This is downstream issue, not an INDI issue per se.

closed 2021-09-28 by knro

JiangXL, 2021-09-28:
Actually, I find that `indi=3rdparty` has `fxload-libusb` and `fxload` modules. @xsnrg How about packaging these?